richardghirst/PiBits #43: Servoblaster not compatible with Raspberry Pi 2

tomfogg, 2015-02-04:
Looks like the DMA addresses have change but I can't find any details on what the new ones are

tomfogg, 2015-02-05:
as some extra info, the userland servod hangs the pi2, the kernel module loads with

`Feb  5 21:18:20 raspberrypi kernel: [    5.821933] ServoBlaster: Control page is at 0xb9364000, cycle_ticks 2000, tick_scale 6, idle_timeout 0`

 but doesnt do anything on the gpio ports

tomfogg, 2015-02-05:
Since it looks like the DMA base has changed to 0x3F000000 from 0x20000000 i tried changing the userland daemon to start from that base but this still hangs the pi unfortunately

Gadgetoid, 2015-02-07:
I have been having this same issue with a WS2812 LED driver which uses DMA and PWM- I have changed the GPIO offsets but suspect there are other changes and a lack of documentation.

popoklopsi, 2015-02-08:
For me it crashes exactly at:
```
for (i = 0; i < num_samples; i++)
    turnoff_mask[i] = maskall;
```

tomfogg, 2015-02-10:
There are some [updates on the rpi forum here](http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99115&start=25) apart from the change in DMA addresses there seem to be some changes in how the cache is being dealt with

vallettea, 2015-02-16:
Also interested in this question, i'm trying to make servoBlaster run on odroid c1, which basically equivalent to rpi2

tomfogg, 2015-02-16:
The [pigpio](http://abyz.co.uk/rpi/pigpio/) library is now working with the pi2, support was added with [this commit](https://github.com/joan2937/pigpio/commit/b14e0a0572346e298ec97f0ee590362f5d59f2d7) I haven't got the understanding of the DMA code to port this fix to servoblaster

richardghirst, 2015-02-17:
That's not a real fix though - as Joan admits in the forum thread the
pigpio code is just accessing other memory locations in the hope of
triggering a cache flush, and initialising the DMA engine multiple times
until it seems to have started up.  I'm hoping we'll come up with a better
solution than that, but I've not got a Pi2 to play with yet.

tomfogg, 2015-02-18:
I thought it looked quite hairy. It does seem to work pretty well though, havent noticed any issues with it controlling 6 servos. If you need some code testing, let me know

richardghirst, 2015-02-19:
Code for testing posted on the forum, see
http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99115&p=699651#p699651

Richard

tomfogg, 2015-02-19:
That code is working a treat for me with 6 servos. Good stuff, especially if you didn't have a pi2 to try it on!

richardghirst, 2015-02-19:
Got my hands on a Pi2 about 5 hours ago :-)

wennycooper, 2015-03-01:
Same problem here. servoBlaster is cool!

richardghirst, 2015-03-01:
Hey Kevin, I assume you saw the link to some working code further up this
thread?

wennycooper, 2015-03-02:
Ahh... yes, you are right. The new code worked on my Pi2.  Thanks a lot!!

luks3n, 2015-05-09:
Hi, today I updated Raspian and also re-compiled ServoBlaster on my Pi B+, and now when I execute a program that used to run just fine, I get "Invalid pin number 33 in P1 pin list", the command I issued to run ServoBlaster is "sudo servod --cycle-time=5000 --step-size=10 --min=1000us --max=2000us --p1pins=13,18,33,40,7 ", really strange. Maybe some pin mapping was changed?

friksa, 2015-09-21:
Any news on an update?

richardghirst, 2015-09-21:
I'm on holiday at the moment so can't look at the code but maybe a problem
with it deciding which version of Pi it thinks it is running on.  Didn't
have 33 pins on the original Pi. You could try to find that code and change
it to think it is always a Pi B+ as a short-term fix.

richardghirst, 2015-10-01:
Oh, I don't have support for the B+ at the moment (should have shortly
though).  I guess if it was working on the B+ for you, you must have used
the servod patch from Martin Habovštiak, see
https://github.com/richardghirst/PiBits/pull/48

luks3n, 2015-10-01:
The fix for Raspberry pi B+ is pretty easy, just add the missing pins to 
```
static uint8_t rev2_p1pin2gpio_map[] = {
	DMY,	// P1-1   3v3
	DMY,	// P1-2   5v
	2,	// P1-3   GPIO 2 (SDA)
	DMY,	// P1-4   5v
	3,	// P1-5   GPIO 3 (SCL)
	DMY,	// P1-6   Ground
	4,	// P1-7   GPIO 4 (GPCLK0)
	14,	// P1-8   GPIO 14 (TXD)
	DMY,	// P1-9   Ground
	15,	// P1-10  GPIO 15 (RXD)
	17,	// P1-11  GPIO 17
	18,	// P1-12  GPIO 18 (PCM_CLK)
	27,	// P1-13  GPIO 27
	DMY,	// P1-14  Ground
	22,	// P1-15  GPIO 22
	23,	// P1-16  GPIO 23
	DMY,	// P1-17  3v3
	24,	// P1-18  GPIO 24
	10,	// P1-19  GPIO 10 (MOSI)
	DMY,	// P1-20  Ground
	9,	// P1-21  GPIO 9 (MISO)
	25,	// P1-22  GPIO 25
	11,	// P1-23  GPIO 11 (SCLK)
	8,	// P1-24  GPIO 8 (CE0)
	DMY,	// P1-25  Ground
	7,	// P1-26  GPIO 7 (CE1)
};
```
there are already some forks of this project which do exactly that.

richardghirst, 2015-10-01:
Except that would let you specify pins on P1 above pin 26 on a rev 2 board,
which is not ideal.  There are 3 configurations of gpio pins: boards with
just P1 with 26 pins, boards with a P1 of 26 pins and a P5 of 8(?) pins,
and boards with just a P1 of 40 pins.

richardghirst, 2015-10-03:
Latest code in github should support Pi B+, Pi2, and 4.1 kernels.

closed 2015-10-03 by richardghirst

improti, 2015-10-04:
Thank you for merging the changes mate, much appreciated! And also for the whole thing, while we are at it! 

I did migrate my code to https://github.com/sarfata/pi-blaster in the meantime, though. As I'm still using servoblaster on several old Pis: Is there the need for these two different projects? Have you guys been in contact and considered merging them into one? To me it seems that pi-blaster has some technical optimisations while servoblaster offers better configuration (via options instead of the need to modify the source and recompile). And it is a fork, so there should be much in common?