richardghirst/PiBits #45: Bus error at service start

vallettea, 2015-02-16:
Hello,

thx for this lib. I'm trying to make servoBlaster work on an odroid c1 which has approximately the same architecture as the rpi 2. The user version compile but I get a 

```
bus error 
``` 

on startup. Do you have any advice on what can cause this ?
Thx

MarkR42, 2015-02-16:
ServoBlaster won't work on anything which is not a Raspberry Pi. It is completely dependent, not just on the CPU architecture, but the DMA controller in particular and specific knowledge of the Pi hardware. It cannot be made to work on odroid c1 or anything else, without particular knowledge of that platform. Servoblaster doesn't use the CPU to wiggle the gpio pins, the DMA controller does it.

vallettea, 2015-02-16:
ok thx

closed 2015-02-16 by vallettea