richardghirst/PiBits #47: dmpPacketBuffer is never initialized

gittymonkey, 2015-03-31:
It appears it was once in the code, but now it is commented out at line 500 in file MPU6050_6Axis_MotionApps20.h

richardghirst, 2015-03-31:
Very true, but then even if that code was included, there is no code
anywhere to write anything in to the allocated buffer that I can see so
allocating the buffer wouldn't really add any value.  As per the README, I
just got this working on the Pi, and that same code is in the original
source I started from.  I have no detailed documentation on the MPU6050,
and have no idea what the original author's intent was.

Do you have an issue that would be solved by including that code at line
500?

gittymonkey, 2015-04-04:
Well no, but it is a potential bug for someone who doesn't know it is not initialized. So why stop at just commenting out and creating a bug, remove all references to it and now you good code. I understand it was there from the original code so you didn't create the issue. I did discover a bigger issue, related to this, and posted it to Jeff's repository in case you are interested.

https://github.com/jrowberg/i2cdevlib/issues/158

Tony