richardghirst/PiBits #49: os.write for /dev/servoblaster.

ProgramErgoSum, 2015-04-24:
Hello!
Python doesn't complain about the following; but, there is no motion of the motors. Is `echo` the only way ?
```
>>> import os
>>> p = os.open('/dev/servoblaster',os.O_WRONLY)
>>> p.write('1=1500us')
>>> os.write(p,'1=1500us')
8
>>> os.close(p)
```

MarkR42, 2015-04-24:
Hi,

I think you need to write a newline character (or use print() ) and
remember to flush the file handle if you leave it open.

This works for me:

servoblaster = open('/dev/servoblaster', 'w')
print >>servoblaster, "%d=%d" % (servo,pulselen)
servoblaster.flush()
# Keep file open to write more commands later.

ProgramErgoSum, 2015-04-24:
Thanks! I appended `\n` and used `flush()` and the following now works for me.

```
>>> p = open('/dev/servoblaster', 'w')
>>> p.write('1=200us\n')
>>> p.flush()
>>> p.close()
>>> p = open('/dev/servoblaster', 'w')
>>> p.write('1=1500us\n')
>>> p.flush()
>>> p.write('1=500us\n')
>>> p.flush()
>>> p.close()
```

MarkR42, 2015-04-24:
It's not necessary to close and reopen the device when you want to send
another command later, you can just keep the file open and write more again
later.

You only need to use flush() if you don't close the device.

ProgramErgoSum, 2015-04-25:
Got it.

The list of commands you see above was done at the Python command line. In my .py script, I have the `close()` only when I have decided to terminate the script.

Happyskull, 2016-05-26:
i haz no skillz in diz area. i haz bad speelings and is nottt gud at pyton. i ned sum srs halp with dis. when i haz start program wit modprobe servoblaster, i it keeps saiyin: Error: could not insert 'servoblaster': Operation not permitted. dis error makes me much sadz and hurtz muh fellings. i haz ned halp.