richardghirst/PiBits #50: Controlling motors in a loop, terminating and intiating a new loop.

ProgramErgoSum, 2015-04-27:
Hi,
I have been able to control motors in an iteration with `servoblaster` successfully. My intention is to loop this iteration infinitely in a process until I send a signal to kill the same. This works too. But, I want a new operation to be triggered after killing this process. At this point, the servo motors don't reset to `500μs` or any other position that I had earlier defined as a default position. When I set the motors individually to the default position from command line, it works, though. 

Can you please tell me how to 'gracefully' terminate the loop (controlling individual motors) and start a loop of different motor positions ?

richardghirst, 2015-04-28:
The basic idea would be to install a signal handler to catch the interrupt
when you try to terminate the loop, and set a flag that the loop checks
every time it gets to 500us. Exactly how you do that would depend on what
language you are programming in.

ProgramErgoSum, 2015-05-02:
Hi Richard,
That is close to what I ended up doing! 

First, I created a pipe with a `mkfifo` that is written to my Python code and read off by `cat` command into `/dev/servoblaster`. This ensured that, `servod` simply sees a stream of commands coming to it without getting involved in looping, termination of looping, new loop etc. Then, in my Python code, I handled the termination of the process to start the next loop.

In general, how does one calculate a sufficient time delay between two writes to the `/dev/servoblaster` ? I am inserting a delay of `100ms`; five times the `20ms` cycle time and it has worked well so far.