richardghirst/PiBits #52: Raspberry Pi 2 B  kernel 4.1 Servoblaster

AFUDIrk, 2015-08-26:
Hello,
I use my Raspberry Pi 2 B with RPi_Cam_Web_Interface where I ansteuere the camera with servos (pan, tilt).
With the update to kernel 4.1, control my servos, which are driven via the servo Blaster no longer works.
Can we expect an update from servo Blaster?

MDB22, 2015-09-07:
I'm also having the same issue, latest RPi update has rendered ServoBlaster unusable.

richardghirst, 2015-09-07:
To make servoblaster work with a 4.1 kernel, you need the latest version
from here:

https://www.raspberrypi.org/forums/view ... 25#p699651
<https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=99115&start=25#p699651>

Then find this line in mailbox.c and change the "100" to "249", run make,
then kill and restart servod:

if (mknod(filename, S_IFCHR|0600, makedev(100, 0)) < 0) {

Worked for me, at least.

AFUDIrk, 2015-09-08:
Habe mir ServoBlaster-20150219.tgz herunter geladen.
In der Datei mailbox.c finde ich die genannte Zeile nicht, um die Änderung vor zu nehmen.

fimholz, 2015-09-08:
Have a look in the other *.c file (servoblaster.c or similar)...

AFUDIrk, 2015-09-08:
In the servod.c I find this entry:
if (mknod(MBFILE, S_IFCHR|0600, makedev(100, 0)) < 0)

Do I change this?

fimholz, 2015-09-08:
Yes, I changed this line to 
if (mknod(MBFILE, S_IFCHR|0600, makedev(249, 0)) < 0) 
and it worked for me ;-)

AFUDIrk, 2015-09-09:
it works!!
many thanks! :-)

isahatipoglu, 2015-09-29:
MDB22,
I am new to downgrading kernel process. Could you please give me some codes to downgrade my kernel version which is 4.1.7-v7+. I have been trying to solve the problem with servoblaster for 7 hours:) 
Thanks

steveedkn, 2015-10-03:
Latest code in github should support Pi B+, Pi2, and 4.1 kernels.

closed 2015-10-03 by richardghirst