richardghirst/PiBits #54: Servoblaster PWMs (+invert not working)

Victor30, 2015-10-21:
Hello! I have a Pi2 and tried to use the latest version of Servoblaster. I'm trying to generate 4 PWMs for controlling 4 ESCs for a quadcopter. I'm using a USBee AX (it's like a small USB oscilloscope) for reading the signals and displaying them using sigrok (open source signal analyzer). I tried reading PWMs generated by other sources (like an Arduino) and the signals looked smooth so I don't think that what I use for reading has a problem. However when I try to read the Servoblaster signals they look like having spikes (on both high and low transitions). Also I started servod with --min=1000us --max=2000us with 20000us period (so 1 to 2 ms pulse with 20 ms period) but the readings look like high ~16ms and then low ~4ms. They don't look like the needed configuration. Also tried the --invert option but in this case no signals appear. (I'm using the userspace version)

richardghirst, 2015-10-21:
Please provide the actual parameters you are passing to servod, the output
servod generates when you start it, the commands you are issuing (echo x >
/dev/servoblaster), and the resulting signal timing you are seeing.

It works for me with a Pi2, so without any specific details of what you are
doing I can't help.

Victor30, 2015-10-22:
[image]

Victor30, 2015-10-22:
ok this is where things get weird...
I uninstalled servod and that same waveform still appears..
also I verified **ALL** pins of RPi2 and all have the same waveform!
even GND and +5V

Victor30, 2015-10-22:
sorry for the disturbance... it seems I had a problem with the grounding of the signal reading tool.. thank you for your support!

closed 2015-10-22 by Victor30