richardghirst/PiBits #55: Servoblaster GPIO-17(P1-11) and GPIO-18(P1-12) not working

Victor30, 2015-10-31:
Any idea why those 2 pins would be blocked? First one is always 0 and second one is always 1.
I have Raspberry Pi2 and I use the latest version of user-space Servoblaster. The other pins work perfectly (even the 8th which I've seen someone has an issue with).

[image]
[image]

Victor30, 2015-11-01:
Also is there a way to write into /dev/servoblaster without using echo? I want to write in it from a C application from a pthread. It seems echo has a problem with this situation. I tried fprintf but nothing happens; the PWM remains unchanged.

MarkR42, 2015-11-01:
You can write to it from a C application, no problem, but be sure to use fflush() after writing (if using fopen) otherwise the data will be buffered inside the C library.

system("echo ...") is an inefficient and needlessly complicated way of doing it.