richardghirst/PiBits #60: How to get current servo position?

timbuktu031, 2016-03-26:
Can I get current servo position? 
I need current position to reverse direction.
advise me.

richardghirst, 2016-03-26:
No, you need to make your code remember what position you have moved it
to.  In the general case someone can force a servo position and there is no
interface to read the actual position from the servo.

calleblyh, 2018-02-22:
I have a similar need to show actual servo position. My idea would be to show the value of "servo1=servo1+2" after each keystroke next to the left, right, up, down which is shown on screen. My servos (pan and tilt) mid position is 150 and limited between 70 and 230. With which syntax can I get the value servo1 shown on screen ? Greatful for any advice how to acheive this !
calleblyh helsinki