richardghirst/PiBits #61: error re-compiling latest Servoblaster

dentex, 2016-03-26:
Hello,
I've been using Servoblaster successfully for ages, in a project of mine involving some "mission critical" applications of PWM and LEDS fixtures :) (https://github.com/dentex/piac).
The initialization command is `sudo servod --p1pins=11,15,19,21,23 --cycle-time=5000us --step-size=5us --min=5us --max 5000us`. It's a B+ Pi model. All good.
Now I was in the needs of using one more PWM pin and I had to use one of the pins the B+ has in addition to the old B model. Namely, I tried the 29. Being an old compiled version not supporting the B+ model (I think), the above command with the pin 29 added gave: `Invalid pin number 29 in P1 pin list`.
So I updated the PiBits source, cd'ed into ./Servoblaster/user, but `make servod` is throwing at me the error:

    gcc -Wall -g -O2 -o servod servod.c mailbox.c -lm
    In file included from /usr/include/arm-linux-gnueabihf/bits/sigcontext.h:28:0,
                 from /usr/include/signal.h:339,
                 from servod.c:35:
    /usr/include/arm-linux-gnueabihf/asm/sigcontext.h:2:8: error: macro names must be identifiers
    /usr/include/arm-linux-gnueabihf/asm/sigcontext.h:4:1: error: unterminated comment
    /usr/include/arm-linux-gnueabihf/asm/sigcontext.h:1:0: error: unterminated #ifndef
    In file included from /usr/include/signal.h:359:0,
                 from servod.c:35:
    /usr/include/arm-linux-gnueabihf/sys/ucontext.h:108:16: error: field ‘uc_mcontext’ has incomplete type
    Makefile:6: set di istruzioni per l'obiettivo "servod" non riuscito
        make: *** [servod] Errore 1

Being a total noob when it comes to source files, C, building and such, now I'm completely clueless on what to do to able to use the other pins from the B+ with Servoblester.

Any hint?
Thank you very much.

richardghirst, 2016-03-26:
That looks to me like sigcontext.h on your system may be corrupt; do you
have another Pi you can try the build on?

dentex, 2016-03-26:
Yes I have it. I can try tomorrow.

I started to think it may be due to my Pi  when I had the same error also
compiling pi-blaster.

If I wanted to fix this file, do you know what's needed to be reinstalled
on the system?

Thanks again.

richardghirst, 2016-03-26:
dpkg -S /usr/include/arm-linux-gnueabihf/asm/sigcontext.h

That should tell you which package provides the file, or your could just
copy the file from another Pi.

dentex, 2016-03-26:
Yesss... Thanks man...

Just did:
sudo apt-get install --reinstall linux-libc-dev
and 'make' went all good. Working again like charm, also for the pin 29!!!

Thank you very much for your help and for the great piece of software
ServoBlaster is.

closed 2016-03-27 by dentex