richardghirst/PiBits #62: ServoBlaster on a Raspberry Pi 3?

emeflag, 2016-04-04:
Richard....  Do you see any issues with ServoBlaster working on a Raspberry Pi  Version: 3?

richardghirst, 2016-04-04:
I don't expect any significant issues, but I don't have one to try.

Richard

emeflag, 2016-04-04:
I’ve got a Raspberry PI 3 on order and will give it a try then.  I’ll let you know one way or the other if it works.

Thanks for the speedy response!  Cheers..

lf-, 2016-06-17:
@emeflag any news on that? I want to find out if it works prior to buying a pi3.

hexaguin, 2016-06-17:
@lf- ServoBlaster works perfectly on my Pi 3.

emeflag, 2016-06-17:
I can confirm servo blaster works fine on a PI3

kuime, 2016-08-27:
system: 2016-05-27-raspbian-jessie
pi model: pi 3 model B v1.2

If a monitor is connected to the HDMI port while the pi boots up, nothing happens.
If the pi boots up without a monitor connection, Servoblaster goes crazy when I run: echo P1-16(any pin)=100(any value) > /dev/servoblaster

When I put the command in rc.local,  I saw the servo turned correctly while the pi boots up, but then if I run the command again in command line later,  the servo turns crazy again.

richardghirst, 2016-09-05:
Have you tried invoking servod with --pcm ?  Someone else had trouble on an
rpi3 and that fixed it for them.

EugeneR1985, 2016-09-22:
Hi Richard! I have problem with my raspi3 and servos. Servo's dont work correctly while the pi boots up. The problem was discussed earlier (post by kuime). How to fix the problem? Thank you!

lf-, 2016-09-22:
Did you try the suggestion of invoking it with --pcm?

EugeneR1985, 2016-09-22:
i create a file in /etc/init.d/ called servoblaster

...
echo "Starting servoblaster"
/home/pi/PiBits/ServoBlaster/user/servod --p1pins="15,16" --pcm
...
right?

kuime, 2016-09-22:
I tried pcm as well,  did not work. Now I have it working perfectly by switching the OS to the lite version  xD.

EugeneR1985, 2016-09-23:
Hi!Use raspbian lite?My servo working correctly with --pcm. Thanks)

Triuman, 2017-01-22:
I'm using SB on Pi3. Before putting an USB audio adaptor, it was working well without --pcm. But while recording sound through USB adaptor, my servo did just small wiggling. Using --pcm solved the problem.

But while daemon is working, I set servo value to 0(zero) to cut the signal as it's in the desired position and "sometimes" the servo does a big move to one side and other times it cuts the power to that servo as desired. As it's the steering servo of a RC car, I can't tell where it really wants to stop since the steering has a max angle.

I'm also using servod to run the brushless motor of the car via ESC(Electronic Speed Controller), so I can't use idle_timeout. How can I stop sending pulses to the steering servo?

AFUDIrk, 2017-02-22:
For me it worked on my Pi 3 without any restrictions.
But be the last update kernel: Bump to 4.9.11, the servos no longer work.
I control the servos via RPi Cam Web Interface.

maxjp, 2017-03-22:
After  updating kernel, the below change needed. it works after the change/compile on my P3.
> cat /proc/cpuinfo |grep -i hardware
> Hardware        : BCM2835

line#960, added the BCM2835 ;-)
>       else if (strstr(modelstr, "BCM2709") || strstr(modelstr, "BCM2835"))

AFUDIrk, 2017-04-11:
else if (strstr(modelstr, "BCM2709") || strstr(modelstr, "BCM2835"))
It works !!
Thank you

Bonzadog, 2017-05-25:
25.5.2017

I have just updated my Rpi2 somewhat old version of jessie and am now getting the Sevmod: cannot..etc
I re installed using the install-pan-tilt.bah  and this did not solve the problem.
But where is the else if (strstr(modelstr, "BCM2709") || strstr(modelstr, "BCM2835")) located?

AFUDIrk, 2017-05-26:
cat /proc/cpuinfo |grep -i hardware
Hardware : BCM2835

In servod.c, line#960, added the BCM2835 ;-)

  else if (strstr(modelstr, "BCM2709") || strstr(modelstr, "BCM2835"))

Bonzadog, 2017-06-07:
I do not have sermod.c on the RPi. When I installed from OpenElectrons the pre-assembled sermod (without sources) was included. 
So I need to find a valid sermod source....

srcshelton, 2017-09-03:
If anyone is still having problems with this, please try the code at https://github.com/srcshelton/servoblaster/ and drop me a line (either here or by opening an issue there) to let me know if anything's improved?

lzto, 2017-10-03:
I can confirm that it does not work on RPI A+ with kernel 4.9.41+
(updated from an older kernel version, and previously it works)
```
coco ~ # uname -a
Linux coco 4.9.41+ #1023 Tue Aug 8 15:47:12 BST 2017 armv6l GNU/Linux
coco ~ # cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 797.66
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0012
Serial		: 0000000065dbe208
coco ~ #
```
servodebug shows the following
```
coco ServoBlaster # ./servodebug
This code should be compiled with the command:

  gcc -Wall -O2 -o servodebug servodebug.c

It should be run with the command:

  sudo chrt 1 ./servodebug

DMA controller is not running - is the module loaded?
coco ServoBlaster #
```

lzto, 2017-10-06:
never mind, I reinstalled the kernel and it works now...
weird though ... previously I have servos connected to P1-38 and P1-40  and they map to servo 0 and 1 respectively. and they works fine. (which is impossible from current version???)
I tried to change them to P1-18 and P1-22 and they does not work initially. but after i reinstalled the kernel and restarted several times it finally works. .. magic....

harlock74, 2017-10-12:
@srcshelton
That did the trick indeed! Could you please explain the different from the original code?

srcshelton, 2017-10-12:
The commit with the changes is [4f42f39](https://github.com/srcshelton/servoblaster/commit/4f42f395e46dfc31ade6bb773be67654a0ed0b58) - and most of these are simple fixes to resolve compiler warnings.
The key change in terms of working on recent hardware, however, is from [line 1025](https://github.com/srcshelton/servoblaster/commit/4f42f395e46dfc31ade6bb773be67654a0ed0b58#diff-364476f388d8ca6306a119734df9fe60R1025) where the (no longer valid) constants used to determine key locations in memory for interfacing with peripherals are replaced with calls to the Broadcom `bcm_host` library - which should provide compatibility across all models of RPi, past and present.

harlock74, 2017-10-13:
@srcshelton
Thanks for that. Very useful to know.

harlock74, 2017-10-13:
@srcshelton
I have just noticed the commands like:
echo 0=+10 > /dev/servoblaster
do not work anymore. 
But commands like:
echo 0=240 > /dev/servoblaster
work perfectly fine. Any thoughts please?
Many thanks!

srcshelton, 2017-10-13:
Hmm, I'm not aware of this being due to anything I've changed - but I'll admit that is the most likely explanation ;)
I'll take a look and see whether I can spot anything suspect!  Sadly, I don't have an RPi handy right now to do real-world tests against...

srcshelton, 2017-10-13:
It's the change around [line 799](https://github.com/srcshelton/servoblaster/commit/4f42f395e46dfc31ade6bb773be67654a0ed0b58#diff-364476f388d8ca6306a119734df9fe60L799) - if you run `servod` directly rather than as a daemon/service, do you see any errors printed about the format of the input when you write the command to the `servoblaster` FIFO?

harlock74, 2017-10-13:
Thanks for helping.
Did you mean run something like:
sudo ./servod ?
If so I don’t get any error messages at all.
Cheers

myklr650, 2017-10-14:
I think what running as a deamon is, You don't need to install to make servod run. You just type sudo make. Then sudo ./servod and it will run. 
 If you install it will boot every time u turn on ur pi.
  I may be wrong tho.

myklr650, 2017-10-14:
I just tested 0=+10% and works fine. I didn't install mine tho. Just 'make' then run

harlock74, 2017-10-14:
Hi,
Thanks for your notes.
If I run only ’sudo make’ then if I runoff example:
sudo echo 0=120 > dev/servoblaster
I get 
/dev/servoblaster: Permission denied

I am using srcshelton fork as the original version doesn’t work at all for me.

Can you please confirm the following steps?

git clone sudo git clone https://github.com/srcshelton/servoblaster.git <https://github.com/srcshelton/servoblaster.git>
cd servo blaster
sudo make

or

sudo git clone https://github.com/richardghirst/PiBits <https://github.com/richardghirst/PiBits>
cd PiBits/ServoBlaster/user
sudo make
sudo make install

Thanks!

harlock74, 2017-10-14:
Ok I started from a fresh image of raspbian:
I can get it to work with the following:

sudo git clone https://github.com/richardghirst/PiBits <https://github.com/richardghirst/PiBits>
cd PiBits/ServoBlaster/user
sudo make
sudo ./servod --pcm

Commands like:

sudo echo 0=140 > /dev/servoblaster

work just fine

but commands like 

sudo echo 0=10 > /dev/servoblaster

don't work.

I have tried adding % like 

sudo echo 0=0% > /dev/servoblaster

but it rotates well above the 10 increments needed.

harlock74, 2017-10-14:
Heads up:
I was wrong, I have just noticed that after running:
`echo 0=+10 > /dev/servoblaster`
I am getting the error message:
`Invalid width '10'`
printed out in a separate terminal where I have run sudo ./servod:

Bonzadog, 2017-10-14:
14.10.2017

I have given up on the Servoblaster and am now using
https://shop.pimoroni.com/products/pan-tilt-hat
and is also good for 3 servo only applications
For more servos,  the adafruit servo module.

All the best

nt

srcshelton, 2017-10-14:
Bear in mind that '`sudo echo 'text' > destination`' won't do what you think: the `echo` command will run with elevated privileges, but the redirection to write to the specified destination will still run as your own user.  Try '`echo 'text' | sudo tee destination >/dev/null`'... which is admittedly a bit of a mouthful, but it does the job ;)

srcshelton, 2017-10-14:
> `Invalid width '+10'`

Ah, this is what I was looking for!  Now let's see how we got there...

srcshelton, 2017-10-14:
(For reference, to drive the four servos supplied with a [MeArm](https://shop.mime.co.uk/collections/frontpage/products/mearm-your-robot-nuka-cola-blue) kit (minus the joystick controller, and using the GPIO pins directly via a breadboard HAT), I was testing with an RPi3 and '`sudo ./servod --idle-timeout=1000 --p1pins=15,16,18,22 --step-size 5 --min 102 --max 444`'.  These latter two values were found by experimentation to find the limits beyond which the servos didn't reliably operate - YMMV.)

srcshelton, 2017-10-14:
... so the error is coming from [line 996](https://github.com/srcshelton/servoblaster/blob/ab67be9a776eba15977a10d97219f530fdaa4643/servod.c#L966) (which should read the same as line 948 - I'll fix that), which means that the `parse_width` function at [line 848](https://github.com/srcshelton/servoblaster/blob/ab67be9a776eba15977a10d97219f530fdaa4643/servod.c#L848) is returning failure.
Ah, got it!  [line 880](https://github.com/srcshelton/servoblaster/blob/ab67be9a776eba15977a10d97219f530fdaa4643/servod.c#L880) is incorrectly using the modified value `digits` rather than the unmodified original `od` - I'll fix that now!

harlock74, 2017-10-14:
Hi Stuart,
That looks very promising! Many thanks for looking at it!
Looking forward to your fix;)

srcshelton, 2017-10-14:
Please try https://github.com/srcshelton/servoblaster/commit/a16934944400b5c290d18911dd77e885a428a4c0

harlock74, 2017-10-14:
Bingo!!!
That worked like a charm Stuart! Thanks a ton for the quick fix! Very much appreciated!!!

srcshelton, 2017-10-14:
😁

japheth12, 2017-12-07:
I am using raspberry pi B rev 1.2, raspbian stretch, with 4.9.59 kernel and echo commands doesn't work but it does not give errors. Please help. Thankyou.

srcshelton, 2017-12-07:
@japheth12 Are you using the code from this repo, or my fork at https://github.com/srcshelton/servoblaster?  If you've not given it a try, then please take a look at this link - I've updated the code from here to use more standardised methods to determine how to hit the hardware, which should increase compatibility.

I've not had chance to test with the latest kernel yet I have to admit - but you're in luck, because I'm planning to move my MeArm from an RPI3 to an old RPi2B (or was it a B+?), so I'll have chance to double-check that everything still works on different hardware then.

harlock74, 2017-12-07:
Hi,
I can confirm that Stuart’s fork works like a charm on my old Raspberry Pi B as well!

japheth12, 2017-12-10:
@srcshelton
I've used your fork but still it doesn't work. Please help.

japheth12, 2017-12-10:
Is it necessary to have "OUT" Mode in the gpio pin? Thanks.

richardghirst, 2018-05-01:
Latest version works fine on Pi3.

closed 2018-05-01 by richardghirst

DilipKumar-code, 2020-07-23:
for people using Raspberry pi 4 add

	else if (strstr(modelstr, "BCM2709") || strstr(modelstr, "BCM2835") || strstr(modelstr,"BCM2711"))