richardghirst/PiBits #66: How to avoid minor jerks in servo motor?

ayosuva, 2016-06-01:
Hi @richardghirst  ,

I am using you servoBlaster library. some time i am getting minor jerks in the servo motor(MG996R Metal Gear Servo Motor).How to avoid these jerks?

richardghirst, 2016-06-01:
I assume you mean the servos are supposed to be stationary, but are moving
a little every now and then.  Check the power to them good; if the power is
not stable they will move.  You could set the idle timeout, so servoblaster
stops driving the servo if it hasn't moved for a couple of seconds.  Then
it should not move at all.

ayosuva, 2016-06-02:
how to set the idle time?

richardghirst, 2016-06-02:
It is in the README.txt here:
https://github.com/richardghirst/PiBits/tree/master/ServoBlaster

Run servod with option --idle--timeout=2000 for example.