richardghirst/PiBits #67: Servo only wants to spin in one direction

jrubinstein, 2016-06-04:
I don't like One Direction much as a band... for servos it's terrible though! 
I've connected up all the servoblaster goodness, and everything seems to work while playing with a smaller hobby servo. Now I've connected by big servo, which is a multi rotation 'sailboat winch' servo, and outputting 650us and 1520us just makes the servo want to spin in the same direction. I have an arduino sketch the spins the servo and 650 and 1520 are the fully open and fully closed positions. Any idea what could be the matter? 

Thanks,
James

jrubinstein, 2016-06-04:
seems like I figured this one out myself. I was powering the servo from a separate power supply. When I moved the +/- DC supply to the +5v and Ground pins on the pi, the servo responds as expected.

richardghirst, 2016-06-04:
Using a separate power supply should be fine but you have to tie the ground
pins of the 2 power supplies together.

closed 2018-05-01 by richardghirst