richardghirst/PiBits #68: Incompatibility with Jessie released in May

harlock74, 2016-06-05:
Hi Richard

I would like to report an issue about the servo blaster software under Raspbian Jessie released in May.

Somehow something in the Servoblaster code gets broken because it doesn't work anymore. However I can confirm that it works fine up to Jessie image released in April (2016.03.18).

When you get a chance would you please be so kind as to confirm any incompatibility issues?

Just for for the sake of giving more info:

uname -r
4.4.12+

Many thanks!

lf-, 2016-06-19:
Odd! I run Arch on my first gen pi and it works brilliantly with the newest kernels out there.

harlock74, 2016-06-19:
Hi If-,
Thanks for your feedback.
It's odd indeed. I have tried with 3 different RPis and still with no luck. Can you please let me know if you are running the latest May Jessie? Just for the sake of it, please find the way I install servoblaster.

wget https://github.com/richardghirst/PiBits/zipball/master
mv master pibits.zip
unzip pibits.zip
cd richardghirst-PiBits-96014c8/ServoBlaster/user
sudo make install 
sudo nano /etc/init.d/servoblaster #to change the timeout from 2000 to 500#
sudo reboot

By the way this methos works fine with the previous Jessie.

Thanks.

lf-, 2016-06-19:
Init system issues? Can you manually start the program? Like /usr/local/sbin/servod.

harlock74, 2016-06-19:
Hi,
I can manually start the program. Problem being either the servos don't move at all or they move slightly and then they get stuck and soon after I can hear a sort of buzzing. I am going nuts;(

richardghirst, 2016-07-14:
Maybe you have something else running that is trying to use some of the
same hardware as servod, and messing up the timing of the servo pulses.
Try stopping servod and restarting it without rebooting - that way if
something else that starts at reboot messes up the hardware, servod might
set it back as it wants it.  Try running it with the --pwm option, if that
doesn't help try --pcm instead.  --pwm and --pcm control what hardware
features servod uses to generate timing.

harlock74, 2016-07-18:
Hi Richard,

Many thanks for getting back to me.

I have tried what you suggested but to no avail. I have tried to re-install the jessie image several times but again no luck. It is odd because if I re-install the March Jessie image servoblaster works as expected. Goa and figure!

As far as I am aware of, I don’t have anything else running. When you get a chance would you please be so kind as to let me know if there is something else that I can check? 

Thanks again.

Kind regards,

Alberto

NelsonBenavides, 2016-08-18:
Hi there,

I'm currently having pretty much the same issue described here.

Servod is set to start automatically on system boot after following instructions.

Regardless of the values I type in to move the servo(s), it only moves to the left. I tried negative values and the servo barely makes it back to the center and the buzzing noise is there constantly.

I tried stopping and starting servod manually as suggested by Richard. After that, the servo(s) work perfectly fine (no more buzzing). If I reboot the Raspberry Pi 3, I would have to stop/start servod manually to make the servo(s) work properly again.

No idea what to do now besides trying to uninstall/re-install ServoBlaster again (?).

Any thoughts?

Using RPi 3, latest Jessie image version, latest kernel version, latest everything else version...

Thanks in advance.

--Nelson.

richardghirst, 2016-08-18:
Clearly in this case something else that runs at startup, and after servod
startup, is messing with the timers or gpio or dma that servod uses.  When
you restart servod it reinitializes all that hardware the way it wants it
configured.

There is a debug utility in the servoblaster github repository; assuming
that utility still works it should help diagnose what the hardware is doing
when servod misbehaves.  IIRC you should start servod with --idle-timeout=0
and set the position of servo 0, then run the debug utility.  I've not used
it myself in ages; if you can make sense of those instructions give it a
go, or let me know and I'll remind myself exactly how to use it.

harlock74, 2016-08-18:
Hi Richard,
This sounds a good idea. I will try to understand how to use this debug tool and will report back as well. I am currently away so I will be able to carry out this test in the next following weeks once I am back home.

Kind regards,

Alberto

NelsonBenavides, 2016-08-19:
Ok Richard. I'll give a try to the debug utility as soon as I can and I'll report back to let you know.

btw, this is a lot of C code written.. awesome! :)

Thank you for your help.

--Nelson.

nyacg, 2016-10-10:
Hi there, did either of you managed to find a solution to this??

I'm running servoblaster fine when I launch it manually but when I try to get it to start from boot (both with the init.d and/or calling the servod program from another file) it does not work.

NelsonBenavides, 2016-10-10:
@nyacg :
Sorry, I haven't been able to get back to this and run the debugger.

Maybe this week I'll have some time.

If you try it and get any results please let us know :)

kologer, 2016-10-14:
Changing the /etc/init.d/servoblaster file to OPTS="--idle-timeout=2000 --pcm"
worked for me.  Raspberry Pi 3 running Raspian Jessie With PIXEL.  I had the same issues with servos as described above.

harlock74, 2016-10-16:
Hi kologer,

Thanks for sharing your findings. That did the trick indeed. I can confirm that everything is working like a charm.

Cheers.

Alberto

ptbw, 2017-03-13:
Just to add another voice. Starting using init.d on a fresh install of Jessie as of this month. Using the default PWM to drive a single micro servo is sluggish and does not respond to simple echo commands to /dev/servoblaster. Using PCM in the init.d opts seems to allow it to function, however manually restarting using init.d and PWM  also works.

NelsonBenavides, 2017-07-08:
Hello all,

Finally a million and half years later, I was able to get back to this project.

Changing the /etc/init.d/servoblaster file to OPTS="--idle-timeout=2000 --pcm" worked as well for me.

Thank you very much!

harlock74, 2017-10-12:
Hi chaps,

Here we go again! I have installed the latest Rasbian Stretch but servoblaster seems kind of broken. I have tried to stop it and manually restart with both -pcm and -pwm option but to no avail.

Could you guys please confirm if it is working for you?

Thanks a lot!

Kind regards,

Alberto

NelsonBenavides, 2017-10-12:
@harlock74 
Oh man, is it broken again?

I have not run any updates on my RPi so I can't confirm, nor deny... sorry.

I guess I'll wait before running any updates.

I'll keep an eye on this threat and thanks for the heads up.

--Nelson.

harlock74, 2017-10-12:
Hi Nelson,

Many thanks for coming back to me.

It’s running fine on my other 3 RPis which are still running an old Jessie image. 

Yep I really regret, I will revert to the old Jessie image ( I think it was may) and for sure I am not going to touch the other 3;)

Regards,

Alberto

myklr650, 2017-10-12:
Hi, had the same problem with latest raspian stretch , but did the 'line 960' fix and works perfect now.

harlock74, 2017-10-12:
Hi, can you please confirm how you fix it?
Many thanks.
Regards,
Alberto

myklr650, 2017-10-12:
https://github.com/richardghirst/PiBits/issues/62 scroll down to maxjp's post . line#960 added BCM2835 . to the served.c file

myklr650, 2017-10-12:
Servod.c*

harlock74, 2017-10-12:
Great, thanks for that. Did you mean that I have to edit that file with
nano and add that line?

myklr650, 2017-10-12:
Yeah, I did it using the vi thing on the command line which was a bit of a faff , but it worked.

harlock74, 2017-10-12:
Great, I will try and come back to you! Thanks again!!

harlock74, 2017-10-12:
Sorry to be of a pain, but could you please confirm where the file is located? thanks :)

harlock74, 2017-10-12:
I have located the file and the line was already there! Anyway I have used the code:
https://github.com/srcshelton/servoblaster/ <https://github.com/srcshelton/servoblaster/>
as advised in the same thread and that did the trick! I am not skilled enough to spot the differences between the two codes but it works:)

@Richard,
When you get a chance would you please advise on that?

Cheers

Alberto

harlock74, 2017-10-14:
Chaps,
Good news! Stuart Shelton fixed the issue with the latest Raspbian Stretch! See the related thread:
[](https://github.com/richardghirst/PiBits/issues/62)