richardghirst/PiBits #75: Pan Tilt Servos moves alone when RPI starts.

rmrbranco, 2016-12-11:
Pan Tilt Servos moves alone when RPI starts.
I do not know if it's a ServoBlaster problem. Is there any way when the RPI starts moving the servos to a fixed location? how to do this?

Regards,
Rui

richardghirst, 2016-12-13:
Servoblaster is normally started from a script in /etc/init.d/.  You can
edit that script to echo commands to /dev/servoblaster once it has started
servod.  That will move the servos to a specific position when the system
boots.

You may get some small movement of the servos at startup because GPIO pins
servod uses are inputs by default, and so are the levels are really
undefined until servod starts and changes them to outputs.

You may also get movement at startup depending on your power arrangement.
If you power the Pi and the servos from different supplies and don't power
them on at exactly the same moment, the servos might see strange levels on
the pins.

Richard

rmrbranco, 2016-12-13:
Understood.
Thank you.

closed 2021-07-02 by richardghirst