richardghirst/PiBits #78: Problem when booting without hdmi display

fabienll, 2017-01-26:
hello,
I am using a raspberry 3 and a pan and tilt camera with two servos.
When I boot with the Pi plugged via hdmi to an external display everything work perfectly, but if I boot without a display then the servos will not move (they only jitter a little bit, just enough to hear).

If I boot with hdmi and unplug hdmi later on it will still work.
If I boot with hdmi connected to a powered off screen it will work.
If I boot without hdmi and plug it later on it will not work.

I tried to play with the hdmi parameters in the config.txt file with no luck.

I also read somewhere that it might be because the Pi is not grounded when not plugged on my display.

Do you have any idea as to what the problem might be?

Many thanks!

laserkestrel, 2017-02-12:
Ive been running my Pi 3 with two servos "headless" for a while. Im now investigating another problem and saw your post. 
Have you checked your PI has enough power, Im running a 2A ipad charger with usb cable to power the PI, or a 7.2v nimh battery pack which both have plenty of power. Hope you get it sorted

fabienll, 2017-02-22:
thanks for your reply.
I am using a 2A 5V power supply which should give me enough power. But to rule this out I powered the servos with a battery and still got the same behaviour without hdmi cable plugged in at startup.
Also I can unplug the hdmi cable after startup and still get the servos to move.
For now it's not really a problem but I wanted to be able to put the Pi anywhere with a battery and this issue limits the Pi's positions near the TV :/

fabienll, 2017-03-07:
I solved the problem by simply using an external battery pack I bought today. Seems to be an issue about the power supply then, not sure what exactly but now I can put the camera anywhere which is what I wanted!