richardghirst/PiBits #79: Start at boot?

cklgr, 2017-02-11:
Hi!

I installed servoblaster following these commands:

_wget https://github.com/richardghirst/PiBits/zipball/master
mv master pibits.zip
unzip pibits.zip
cd richardghirst-PiBits-96014c8/ServoBlaster/user
sudo make install
sudo nano /etc/init.d/servoblaster #to change the timeout from 2000 to 500#
sudo reboot_

But I do not get it to work properly until I start it from console after boot.

I've tried adding a bash file to both /etc/init.d and crontab

```
#!/bin/bash
### BEGIN INIT INFO
# Provides:          ServoBlaster
# Required-Start:    $syslog
# Required-Stop:     $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: blabla
# Description:
#
### END INIT INFO

sleep 32
cd richardghirst-PiBits-96014c8/ServoBlaster/user
sudo ./servod --idle-timeout=500
cd
```

But I still do not get it to work unless I run the commands manually.
Any help please?

salk31, 2017-03-06:
Which OS do you have? Sadly it slightly matters what version (some use systemd).

Even pre-systemd you need links from the rc? folders to your /etc/init.d file.

cklgr, 2017-03-06:
I'm using raspbian, 4.4.50-v7+ now.

salk31, 2017-03-06:
OK I think that is based around systemd so I'd suggest going that route.

So /etc/init.d is considered old style.

Somebody has already created a file https://github.com/richardghirst/PiBits/pull/76/commits/92d1eb71638666348bd46d30cd59ad41412a3b1d

It is pretty simple. Mine is:
[Unit]
Description=servo blaster

[Service]
ExecStart=/home/pi/tank/bin/servod --p1pins=29,31,11
Type=forking

[Install]
WantedBy=multi-user.target

Lots of instructions on the web about where to put this file and tell systemd to use it. e.g. http://www.raspberrypi-spy.co.uk/2015/10/how-to-autorun-a-python-script-on-boot-using-systemd/

Make any sense?

cklgr, 2017-03-06:
Thanks a lot for your quick response, I have reinstalled in the middle and it seems to work, at least from the console, I have to reinstall the rpi web cam module (it's the one I want to make it work) and see, but for now I run out of time.
Tomorrow I will continue with it, if I fail I prove your solution.

Thanks again.