richardghirst/PiBits #82: Servod : Cannot parse the hardware name string... :(

Pilou72, 2017-03-14:
Hello,

I had ServoBlaster working on my previous Raspbian (before formating the microSD...), but I don't get to make the installation again.
I did :
sudo git clone https://github.com/richardghirst/PiBits
cd PiBits/ServoBlaster/user
sudo make
sudo make install

And then sudo ./servod to see the servo mapping info, but I get "Servod : Cannot parse the hardware name string"

Any help please ??

Thks
Pierre

dmantz, 2017-03-19:
I did the same and had the same results.
When I try to give a command
echo 0=120 > dev/servoblaster
I get the answer
Access dinied.
Any help please ?

sliksoft, 2017-03-20:
Hi I have the same problems with a rasp **zero** 
sudo git clone https://github.com/richardghirst/PiBits
cd PiBits/ServoBlaster/user
sudo make
sudo make install

And then sudo ./servod to see the servo mapping info, but I get "Servod : Cannot parse the hardware name string"

kernel Linux raspberrypi 4.9.16+

dmantz, 2017-03-22:
Hi, finally I did what describes maxjp in #62 
in the file PiBits/ServoBlaster/user/servod.c 
after that 
sudo make
sudo make install
and now it works.
Thanks maxjp .

Pilou72, 2017-03-22:
Great, that works for me too !! Thanks you so much ! 
I didn't think of using the powerful "grep" as maxjp did to find the origin of the error code...

Cheers !

NewtownGal, 2017-08-18:
I made the change to servod.c and recompiled on a Pi-Zero running Raspian Wheezy. I no longer get a hardware name error. However, when I query /etc/init.d/servoblaster status, it says ERROR: DMA not running. I rebooted with no change. How do I fix this ?

richardghirst, 2017-08-23:
Can you do the following from the directory where you built servod, and
send me the output you get?

cat /proc/version
sudo killall servod
sudo ./servod
echo 0=50 > /dev/servoblaster
echo status /dev/stdout > /dev/servoblaster
echo debug > /dev/servoblaster
ps ax

That last one is to show me what else you have running; maybe something
else on the system is interfering with it.

The change I made to servod to make it work on Pi-Zero for me was:

diff --git a/ServoBlaster/user/servod.c b/ServoBlaster/user/servod.c
index 341776c..f0c0e74 100644
--- a/ServoBlaster/user/servod.c
+++ b/ServoBlaster/user/servod.c
@@ -959,6 +959,8 @@ get_model_and_revision(void)
                board_model = 1;
        else if (strstr(modelstr, "BCM2709"))
                board_model = 2;
+       else if (strstr(modelstr, "BCM2835"))
+               board_model = 1;
        else
                fatal("servod: Cannot parse the hardware name string\n");

Is that what you did?

Richard

NewtownGal, 2017-08-23:
At 07:14 PM 8/23/2017 +0000, you wrote:
Can you do the following from the directory
where you built servod, and
send me the output you get?

I included unsuccessful attempts to use git and diff.

cat /proc/version
sudo killall servod
sudo ./servod
echo 0=50 > /dev/servoblaster
echo status /dev/stdout > /dev/servoblaster
echo debug > /dev/servoblaster
ps ax

Please do not publish this console text. 

The cpu is 97% idle when I do this.

root@Zero-1:~# diff --git
a/ServoBlaster/user/servod.c b/ServoBlaster/user/servod.c
diff: unrecognized option '--git'
diff: Try `diff --help' for more information.
root@Zero-1:~# apt-get install git
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version.
git set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
root@Zero-1:~# diff --git a/ServoBlaster/user/servod.c
b/ServoBlaster/user/servod.c
diff: unrecognized option '--git'
diff: Try `diff --help' for more information.
root@Zero-1:~# apt-get install diff
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package diff is not available, but is referred to by another
package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  diffutils

E: Package 'diff' has no installation candidate
root@Zero-1:~# apt-get install diffutils
Reading package lists... Done
Building dependency tree
Reading state information... Done
diffutils is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
root@Zero-1:~# diff --git a/ServoBlaster/user/servod.c
b/ServoBlaster/user/servod.c
diff: unrecognized option '--git'
diff: Try `diff --help' for more information.
root@Zero-1:~# ls -all servo*
-rwxr-xr-x 1 pi pi   879 Feb  3  2014
servoblaster.sh
-rwxr-xr-x 1 pi pi 48723 Feb  3  2014 servod
root@Zero-1:~# cd servod
-bash: cd: servod: Not a directory
root@Zero-1:~# cd ServoBlaster_Github_Oct2015_Rev_Aug2017
root@Zero-1:~/ServoBlaster_Github_Oct2015_Rev_Aug2017# ls -all
total 236
drwxr-xr-x  2 root root  4096 Aug 18 13:05 .
drwxr-xr-x 17 root root  4096 Aug 23 15:17 ..
-rw-r--r--  1 root root  1207 Feb 11  2016
init-script
-rw-r--r--  1 root root  7602 Feb 11  2016 mailbox.c
[... 429 lines cut ...]
the thread.

Steve Morton

Boundless Security Systems, Inc. (BSSI)
low power outdoor wireless covert
cameras
261 South Main Street, Unit #285; Newtown, CT 06470
USA
tel. 203-445-0562, fax 203-304-2048
Skype: BoundlessSecurity.com
smorton@BoundlessSecurity.com,
www.BoundlessSecurity.com

Boundless Security Systems, Inc., Confidential
& Proprietary Statement: This email and any attachments are
confidential and may be legally privileged. If you are not the intended
recipient, or believe that you have received this communication in error,
please do not read, print, copy, re-transmit, disseminate, or otherwise
use the information. Please delete this email and any attachments,
without reading, printing, copying, forwarding or saving them and notify
the sender immediately by reply e-mail. No confidentiality or privilege
is waived or lost by any transmission in error. Boundless Security
Systems, Inc., and its affiliates reserve the right to monitor all e-mail
communications through its networks. Thank you.

richardghirst, 2017-08-23:
> tel. 203-445-0562 <(203)%20445-0562>, fax 203-304-2048 <(203)%20304-2048>

Your change to servod.c makes the BCM2835 look like board_model 2, whereas
we need it to look like board_model 1.

I've attached my servod.c for you to try.

BTW, you said "Please do not publish this console text.", but this thread
is all on github for everyone to see :)  richardghirst@gmail.com if you
want to go private.

Richard

richardghirst, 2017-08-23:
Doesn't look like you can add attachments to github comments, so I've
pushed my change up to github.  You can get the file here:

https://github.com/richardghirst/PiBits/raw/master/ServoBlaster/user/servod.c

Richard

srcshelton, 2017-09-03:
It looks to me as if `board_model = 1` is incorrect, and the logic elsewhere in the code only works when `board_model = 2` for other devices... but I could be missing something ;)

Please could you try the code at https://github.com/srcshelton/servoblaster/ and drop me a line (either here or by opening an issue there) to let me know if anything's improved?

shahrzadfeghhi, 2021-06-28:
Hi all, so I am trying to make ServoBlaster work for me. but I get the error can not parse the harware name string and permission denied. I don't also know how to run PCM(pulse code modulation). Would appreciate any guidance

richardghirst, 2021-07-02:
@shahrzadfeghhi  please try the latest version of servoblaster (I updated it yesterday), and open a new issue if you still have problems.

closed 2021-07-02 by richardghirst