richardghirst/PiBits #84: ServoBlaster not working on Raspberry Pi Zero W

ileamihai, 2017-04-22:
Hi Guys,

I am struggling with a pan/tilt project on a RPI Zero W.

I am following Allan H's Robot Project from here: https://www.youtube.com/watch?v=wzOkuglA_5s

ServoBlaster was installed as follows:
- git clone https://github.com/richardghirst/PiBits.git
- cd /PitBits/ServoBlaster/user
- changed the servod.c file in order to fix the hardware issue from [here](https://github.com/richardghirst/PiBits/issues/62). maxjp's post from 22 March.
- sudo make servod
- sudo make install
- sudo ./servod
echo P1-12=120 > /dev/servoblaster (NOT working -> PAN servo on pin 12)
echo P1-16=70 > /dev/servoblaster  (NOT working -> TILT servo on pin 16)

I've been scouring the Internet for a solution but cannot find anything...

Everything is up-to-date as of today (RPI-update, update, upgade).

Thanks a lot for your feedback!

githubpac, 2017-05-17:
I had the same issue and fixed it with 2 atempts to edit  servod.c  to change the Board Model to either 1 or 2.   Edit line 960  per Maxjp response on https://github.com/richardghirst/PiBits/issues/62.  I first added || strstr(modelstr, "BCM2835") to  the "else if " line to force a Board Model 1, but that didn't work so I removed and added it to if line just above and  Made it a Board Model 2

ileamihai, 2017-06-05:
Not working... changed the lines as you said, recompiled, reboot, still no movement... and its not a wiring issue (checked that).
Maybe I need to try a clean install...

tambechristian, 2017-06-24:
Hi ileamihai. I have the same issue and I'm also following Allan H project but with Raspberry Pi W zero, but the pan-tilt camera is not responding to the servoblaster. 
Has anyone figured this out?
Thanks

priema, 2017-07-08:
I also have the same problem. :(
Does anyone already know how to fix it?

neutralvibes, 2017-08-02:
Unfortunately I have spent considerable time trying various things, including the fix mentioned above & downgrading the firmware, with no results. It appears it may require an older version of the OS, which isn't really a long term solution. I hope some great servoblaster guru can help to dull the pain a 
 growing numbers of users are feeling, at not being able to use such a great utility.

:(

MarkR42, 2017-08-02:
I think that probably Servoblaster has had its time. Now Pigpio is much better supported (e.g. on newer OS and chip versions) and supports a superset of the functionality, these users are probably advised to use Pigpio instead.

NewtownGal, 2017-08-20:
If no one can figure out how to make the beloved Servoblaster work on Pi-Zero, can someone tell us how to configure Pigpio to act as a drop-in replacement for Servoblaster ? Also, can Pigpio give 2 uS resolution like Servoblaster ?

richardghirst, 2017-08-23:
Sorry I've been ignoring you for so long... had a lot of real life to deal
with lately.

I've just made this edit to servod.c and it seems to be working for me on a
Pi Zero:

diff --git a/ServoBlaster/user/servod.c b/ServoBlaster/user/servod.c
index 341776c..f0c0e74 100644
--- a/ServoBlaster/user/servod.c
+++ b/ServoBlaster/user/servod.c
@@ -959,6 +959,8 @@ get_model_and_revision(void)
                board_model = 1;
        else if (strstr(modelstr, "BCM2709"))
                board_model = 2;
+       else if (strstr(modelstr, "BCM2835"))
+               board_model = 1;
        else
                fatal("servod: Cannot parse the hardware name string\n");

At least, after "echo 0=50 > /dev/servoblaster" a scope on P1-7 shows a
regular 500us pulse every 20ms, as expected.  I don't actually have it
wired to a servo yet.

Please let me know if that helps you guys.

Richard

tambechristian, 2017-08-23:
It works Richard ! I just tested in on a Pi Zero W with two servos. Thank you !!

salmanarnab, 2017-10-19:
hello, i tried on raspberry pi model 3B. my servo is not responding. can anyone give me any suggestion??

RAHULM44, 2017-11-28:
Hey I am also having same problem with raspberry pi 3 Model B. If anyone found solution then please post it. Thanks

srcshelton, 2017-12-07:
Hi all, I've not had the chance to test on RPiZ hardware, but could you try the fork at https://github.com/srcshelton/servoblaster (specifically the change at See changes at https://github.com/srcshelton/servoblaster/commit/4f42f395e46dfc31ade6bb773be67654a0ed0b58#diff-364476f388d8ca6306a119734df9fe60R1025) and confirm whether this is also effective on the Zero (which I believe it should be!)

richardghirst, 2018-05-01:
Relevant fixes for PI Zero and Pi3 copied from from Stuart's version to this repo.  Closing.

closed 2018-05-01 by richardghirst