richardghirst/PiBits #88: Servos don't move

Giudalpa, 2017-06-09:
Hi,
I've tried to install Servoblaster to move 2 servos but without success. I installed it according the typical installation instructions (i tried changing the 960th line as well) and when i put the echo commands, nothing happened, they don't move (no error, just nothing). It's weird, because I made a code in python and both worked, so isn't a problem of wire conections.
I'm using SG90 servos (GPIO17 and GPIO4) and my raspberry is the model 1 B (Rev 000e), with Raspbian Jessie FW 4.9.24+

Hope someone can help me (i'm no a expert).
Thanks in advance.

ianmaynard, 2017-06-13:
I got the same problem. I tried to test run my servos one at a time, just to make sure ServoBlaster and my servos are good, but yeah, i got the same problem as Giudalpa's. Nothing happened when I used echo command, I tried to use PCM too but nothing happens. I also tried to run my servo with square wave generator, and it works fine. I'm using raspberry pi 3 with raspbian jessie.

Can someone help please.

Giudalpa, 2017-06-13:
So I finally made it, I could move both servos! Just use an older version of the OS (I used Jessie LITE 4.4.11+).

richardghirst, 2018-05-01:
Latest version works with recent kernels.

closed 2018-05-01 by richardghirst