richardghirst/PiBits #90: Servos acting strange

devekrehbiel, 2017-07-27:
I am obviously doing something wrong. The servos DO respond to the pins P1=37 and P1=40 on the Pi3. When I initiate the echo commands I get some clicking, maybe sometimes it will cycle from one end to the other of their travel, but nothing that resembles proper behavior. Here is what I have done so far:

1) Changed power supplies. I have a bench unit I tried everything from 5 volt to 9 volts but the clicking sounds the servos make does not get stronger or weaker. The bench supply I am using strictly for the servos is a 50 amp supply.

2) I ran a ground wire from the Pi to the bench supply just to make sure they were both at the same ground potential. No change. The Pi runs on its own 3amp supply at 5v.

I am using the instructions found here about midway down to install ServoBlaster:
http://www.instructables.com/id/IoT-Raspberry-Pi-Video-Streamer-and-PanTilt-Camera/

To try the servos I am using echo P1-40=50% >/dev/servoblaster and variations thereof to get the servos to move. When I hit enter after the command, weak clicking inside the servo.

When you first load up Raspbian, you need to turn on SSH for it to work on a terminal.. also there is GPIO on or off, Wire One on/off, Serial on/off. Do any of those things make a difference? Any thoughts as to what I am doing wrong? Thank You!

devekrehbiel, 2017-07-29:
Ok, after nobody bothering to comment for a week, I have boiled it down to the Pi 3 needing to be downgraded to last Mays version of Jessie? Then there is a bunch of talk about how to go about this, but as an amateur I have no idea how to make that happen without specific instructions. Then, you can't do a sudo upgrade because then you are up to date again which nullifies the older version? Isnt there more to the upgrade than just the Kernel? I like to keep things up to date, but you are telling me that I need to go backwards? Anyone? How is this done exactly? I really do not want to add an Arduino just for pan and tilt operations.

superddr, 2017-09-02:
I believe that you just need a stronger power supply. 5V and 2A for Pi will be fine.

devekrehbiel, 2017-09-02:
Hi Richard. No it wasnt the power supply, it was whatever the newest flavor
of raspbian for the Pi is. Its interfering with your code and anyone
wanting to use ServoBlaster has to go backwards in time to make it work. I
hope you have time to take a look at this, but if you have moved on its
okay, I went with an Arduino to move the servos.  Its not ideal, but it
works. It IS a shame though. ServoBlaster was the salvation of many of the
do-it-yourself robotics and project people out there.

I cant imagine what it must be like for you. They keep screwing
up/"improving" the release OS and then in their changes, something else
breaks. Then you are the slave that has to keep upgrading your software or
it goes obsolete. Hang in there. I really think ServoBlaster was a good
idea and still is.

Deve

richardghirst, 2018-05-01:
Closing this as Deve has it working with newer s/w now.

closed 2018-05-01 by richardghirst