richardghirst/PiBits #91: Raspberry Pi Zero Kernel 4.9 Servoblaster

AFUDIrk, 2017-08-08:
Hello Richard,
I use my Raspberry Pi Zerro with RPi_Cam_Web_Interface where I ansteuere the camera with servos (pan, tilt).
With the update to kernel 4.9, control my servos, which are driven via the Servoblaster no longer works.
Can we expect an update from Servoblaster?

richardghirst, 2017-08-23:
With the patch I just posted on issue #84 to make it recognise the BCM2835
as a board_model 1, servod seems to work for me on a Pi Zero.

I'm running this kernel:

pi@raspberrypi:~/PiBits/ServoBlaster/user $ cat /proc/version
Linux version 4.9.35+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG
crosstool-ng-1.22.0-88-g8460611) ) #1014 Fri Jun 30 14:34:49 BST 2017

AFUDIrk, 2017-08-25:
Hi Richard,
Had made the change yesterday, now the servos function again.
Thank you very much!!!

srcshelton, 2017-12-07:
Hi @AFUDIrk, I've not had the chance to test on RPiZ hardware, but could you try the fork at https://github.com/srcshelton/servoblaster (specifically the change at See changes at https://github.com/srcshelton/servoblaster/commit/4f42f395e46dfc31ade6bb773be67654a0ed0b58#diff-364476f388d8ca6306a119734df9fe60R1025) and confirm whether this is also effective on the Zero (which I believe it should be!)

AFUDIrk, 2017-12-10:
> See changes at srcshelton/servoblaster@4f42f39#diff-
> 364476f388d8ca6306a119734df9fe60R1025

Can test it on occasion, but can take a while. What is different about this
version than on your servo blaster? I am not so good at programming and I
have no experience in programming.

srcshelton, 2017-12-12:
@AFUDIrk the entirely non-technical answer is that the updated version does the same thing, but better ;)
The slightly technical version is that the Raspberry Pi foundation released a new library after the original ServoBlaster code was written to make direct hardware access more consistent across different RPi models, and so I've been able to make use of this to make the SB code safer.  Also, model detection in the original code had got a bit confused, and I've corrected this.
Hope this helps!

AFUDIrk, 2017-12-15:
Did it today. Unfortunately, your servoblster is not. The servo makes only
a short sound, so it is controlled, but it does not move. Have your version
removed from the servo blaster and then installed the servo blaster from
Richard, then the servo works.

calleblyh, 2018-02-18:
I have made numerous attemps to install PiBits/ServoBlaster on my RPi3B without success. I am trying to using arrowkeys to control pantilt servos according to pantilt.py fromTuckerShannon. Everything else works except servos are not moving. I am running on latest Stretch release 2017-11-29. uname -r  4.9.59-v7. Now I wonder is there a compatibility problem ? After reading numerous postings on different forums it is also slightly unclear to me should PiBits/ServoBlaster be installed in home/pi directory or somewhere else like cd / ?. I am totally confused after five days of struggling...
calleblyh helsinki finland

AFUDIrk, 2018-02-19:
A​fter updating kernel, the below change needed. it works after the
change/compile on my P3.

Install Servoblaster:

git clone https://github.com/richardghirst/PiBits.git

cd /PiBits/ServoBlaster/user

​*In servod.c, ​line#960, added the BCM2835*

*ORIGINAL:     else if (strstr(modelstr, "BCM2709"))  *

*modification: else if (strstr(modelstr, "BCM2709") || strstr(modelstr,
"BCM2835"))*

sudo make

sudo make install

finished, ServoBlaster it works.

calleblyh, 2018-02-19:
Heureka,it works... I am really happy I found this posting
Thank you ever so much

srcshelton, 2018-02-19:
(Just a quick note to mention that my fork's had this change for the past six months: https://github.com/srcshelton/servoblaster/blame/master/servod.c#L1019 - I mention this only in the spirit of giving people another option to try if things aren't working for them!)

skhalozulu, 2018-04-30:
@AFUDIrk  It is working with these '' echo 1=10% > /dev/servoblaster '' but not interfacing with the RPi web cam interface. when I touch on the web nothing happens but when I do the above commands the servos move

richardghirst, 2018-05-01:
Relevant fixes copied from from Stuart's version to this repo.  Closing.

closed 2018-05-01 by richardghirst

richardghirst, 2018-05-01:
@skhalozulu your latest problem seems to be with the RPi web cam interface, rather than with servoblaster , so I closed this.  Feel free to open a separate issue for the web cam interface, if you cannot find somewhere more appropriate to ask.  You could try looking in /var/log/apache2/error.log to see if there are any clues.  I know getting that web cam stuff working on a recent Raspbian release can be a bit painful.

skhalozulu, 2018-05-01:
@richardghirst Thank you sir I solved the problem, everything is working smoothly now.