richardghirst/PiBits #94: Servoblaster interfering with NeoPixel rpi_ws281x library

prutsky, 2017-09-17:
I run one servo on a Pi Zero (v1.3) on GPIO4.

For my project I also installed a NeoPixel stick. This uses GPIO18. How can I make Servoblaster not interfere with the NeoPixel GPIO?
As soon as I make install Servoblaster, the NeoPixel does not work anymore as it should. When I uninstall ServoBlaster the NeoPixel LEDs work as they should.
Where can I tell ServoBlaster to not use GPIO18, or is this kind of exclusion not possible?

amigatommy, 2017-10-21:
In /etc/init.d/servoblaster:
hang on "--p1pins=18" at the line beginning with "OPTS="
This tells servoblaster to use GPIO18 as channel 0.
Now you can set servo 0 on GPIO18 to a pulse width of e.g. 1.2ms at the shell prompt:
echo 0=120 > /dev/servoblaster

amigatommy, 2017-10-21:
It should work, but i have nearly the same problem with onewire on GPIO4. I defined only
GPIO11 in the /etc/init.d/servoblaster, but onewire dont work anymore, when servoblaster is active.

srcshelton, 2017-12-07:
Apologies, I've only just seen this - please see take a look at my reply at https://github.com/srcshelton/servoblaster/issues/1