richardghirst/PiBits #96: Survoblaster is not working on Pi Model 3B

salmanarnab, 2017-10-19:
please help me. my raspberry pi model 3b is not responding to echo command . how can i fix my issue?

amigatommy, 2017-10-21:
try this : [https://github.com/srcshelton/servoblaster/]
it works on my RPi 3 !

salmanarnab, 2017-10-26:
i installed but it said permission denied when i try " echo p1-12=120 >/dev/servoblaster"

please suggest me.

amigatommy, 2017-10-26:
Ok, try "sudo echo p1-12=120 > /dev/servoblaster".

salmanarnab, 2017-10-27:
there is another problem .as soon as I install the servoblaster it affecting other gpio pin.

salmanarnab, 2017-10-27:
thanks Amigatommy. it works now.

amigatommy, 2017-10-27:
with pleasure ;o)

ygowill, 2017-11-01:
i install it and use"sudo make servod".but when i use "./servod" it said can't open /dev/men:permission denied, so i try to use "sudo ./servod" and the terminal seems get stuck there

amigatommy, 2017-11-05:
In which folder you have installed ? I recommend to install servoblaster within your home folder, there you have no problems with any permissions. Something is wrong at your installation, what is the path  /dev/men ?

archy587, 2018-01-04:
@ygowill 

I had same issue with permission denied and hanging command prompt. I was working over SSH so I got it working using sudo raspi-config and making sure GPIO Server was enabled under advanced options. then launched servod using:

sudo nohup ./servod & 

to prevent hangup. normal echo commands worked after that...

richardghirst, 2018-05-01:
Relevant fixes copied from from Stuart's version to this repo.  Closing.

closed 2018-05-01 by richardghirst