richardghirst/PiBits #99: Contribution to ServoBlaster for Pi-3B

hkeown, 2017-12-03:
Great driver, spent several days trying to get the driver to work. Finally used the driver from  https://github.com/srcshelton. I've documented the various steps so that any future user of ServBlaster can save time and effort.

[Pi3 ServoBlaster (02122017).pdf](https://github.com/richardghirst/PiBits/files/1524970/Pi3.ServoBlaster.02122017.pdf)

srcshelton, 2017-12-07:
Many thanks for that!

I should document this explicitly, but I moved `/dev/servoblaster-cfg` to `/var/run/servoblaster.status`, which felt like  more standard location - however, you can change this to wherever you like by setting 'STATUSFILE' in the supplied init script (and this defaults to `/tmp/servoblaster-status`).

srcshelton, 2017-12-07:
(There was also a comment about changing `/etc/modules` - this would only have affected the older kernel-module variant of servoblaster, and so is no longer applicable.  The idle timeout can be set from the init script)

radeekhan, 2017-12-16:
When i am running this command ''sudo chrt 1 ./servodebug'' getting this output:
DMA controller is not running - is the module loaded?

Please help me.

richardghirst, 2018-05-01:
Relevant fixes copied from from Stuart's version to this repo.  Closing.

closed 2018-05-01 by richardghirst