richardghirst/PiBits #101: How do you uninstall Servoblaster

Devdroid2, 2017-12-09:
Please,can you tell me how to uninstall everything that came with the servoblaster?

Devdroid2, 2017-12-09:
> Hi,Yes, I need to clean this up.If you ran "make install" then you installed the servoblaster.kokernel module too (see the README, there is the old kernel module,servoblaster.ko, that is installed via 'make install' and the newer userspace servod.c for which you just 'make' and not 'make install').Anyway, to stop the system loading servoblaster.ko, you can:rm /lib/udev/servoblasterrm /etc/udev/rules.d/20-servoblaster.rulesand then reboot.Following that, you can still run your modified servod, and the camerashould work.Richard
>
> Because those files dont exist

I have allready tried what you said on the other post like this and it doesnt work:

Hi,Yes, I need to clean this up.If you ran "make install" then you installed the servoblaster.kokernel module too (see the README, there is the old kernel module,servoblaster.ko, that is installed via 'make install' and the newer userspace servod.c for which you just 'make' and not 'make install').Anyway, to stop the system loading servoblaster.ko, you can:rm /lib/udev/servoblasterrm /etc/udev/rules.d/20-servoblaster.rulesand then reboot.Following that, you can still run your modified servod, and the camerashould work.Richard

richardghirst, 2018-05-01:
"make install", "make uninstall" works these days.

closed 2018-05-01 by richardghirst