richardghirst/PiBits #102: bash: /dev/servoblaster: Permission denied

thunderfangepic, 2017-12-11:
Hello, I am a newer pi user, and I am having some trouble using the ServoBlaster. I have already run the make servod commands, and am trying to control the servo using 

echo P1-18=120 > /dev/servoblaster

But I am getting this error right away:

bash: /dev/servoblaster: Permission denied

If anyone could help I would very much apreciate it. Thanks!

amigatommy, 2017-12-11:
Hi, you must write: 
sudo echo P1-18=120 > /dev/servoblaster
Greetings

srcshelton, 2017-12-12:
Redirection doesn't work like that :(
Try:
`echo 'P1-18=120' | sudo tee /dev/servoblaster >/dev/null`

thunderfangepic, 2017-12-13:
Hi, thanks for the help, but when I type `echo 'P1-18=120' | sudo tee /dev/servoblaster >/dev/null` it doesn't take the argument as complete. It gives me a new line to enter more. Am I doing something wrong?

srcshelton, 2017-12-13:
@thunderfangepic Did you type the above, or copy/paste?  Make sure nothing is changing the `'` from your keyboard into smart-quotes which look nice but aren't actually understood by a shell!  Trying the above but with `"` double-quotes may also help...

thunderfangepic, 2018-01-08:
It worked, Thanks!

closed 2018-01-08 by thunderfangepic