richardghirst/PiBits #106: How to change default pins on ServoBlaster?

Mtousi, 2018-03-29:
Hi guys, i have been working on a project that requires at least 4 channels of PWM to control 4 servo motors. but the first 26 set of pins on my raspberry pi P1 header are taken by the touche screen LCD. therefore i need to change the default pins from (7,11,12,13,15,16,18,22) to (31,32,33,35,36,37,38,40).
I have been trying to use "sudo ./servod --p1pins=31,32,33,35,36,37,38,40" and the servo mapping seems to be set and changed, but as soon as i look back on the properties i notice that all the pins are back to default and nothing seems to be written.

I'm pretty new to this and not quite sure how to tackle this issue! Any suggestions?

myklr650, 2018-03-29:
Personally, I run a start up script that  starts servod with my preferred settings.

haoudar, 2018-04-03:
Hi mtousi, you can map pins by editing servod.c then make ... and so on

richardghirst, 2018-05-01:
I don't know what you mean by "look back on the properties".  Are you still having issues?

richardghirst, 2021-07-02:
Closing as no recent activity.

closed 2021-07-02 by richardghirst