richardghirst/PiBits #112: undefined reference to `makedev' -> add sys/sysmacros.h

Serveurperso, 2019-07-22:
inside mailbox.c we must add :

#include <sys/sysmacros.h>

ROZBEH, 2019-09-10:
The make was successful on PI4 but when I enter the command ```sudo 2=200 > /dev/servoblaster```, the linear actuator does not move at all. I'm sure that the linear actuator is connected to 2. I tried it with PI3 and it works but it does not work on PI4. Helps is appreciated.

richardghirst, 2019-09-10:
I don't have a Pi 4 yet, and am away from home for a couple of weeks.  I'll
get one and figure it out when I'm back, unless someone beats me to it.

Richard

richardghirst, 2019-10-01:
> Thanks. Much appreciated

I've just pushed updates to github; it should work on Pi 4 now.

ROZBEH, 2019-10-02:
Thanks so very much.

closed 2021-07-02 by richardghirst