richardghirst/PiBits #113: Catastrophic Segfault on PI 4 (crash-loop the ssh server forever)

Serveurperso, 2019-07-22:
building on Raspberry PI 4 kill it. It's systematic
We can't connect to ssh anymore.

root@raspberrypi:~/PiBits/ServoBlaster/user# make install
[ "`id -u`" = "0" ] || { echo "Must be run as root"; exit 1; }
cp -f servod /usr/local/sbin
cp -f init-script /etc/init.d/servoblaster
chmod 755 /etc/init.d/servoblaster
update-rc.d servoblaster defaults 92 08
/etc/init.d/servoblaster start
make: *** [Makefile:14: install] Segmentation fault
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.709726] Internal error: Oops: 207 [#1] SMP ARM
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710501] Process servoblaster (pid: 5453, stack limit = 0x1b25780d)
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710524] Stack: (0xdaab9d7c to 0xdaaba000)
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710550] 9d60:                                                                be90d088
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710590] 9d80: 00030000 00000000 be90d17c daab80f8 daab8000 c02045e4 c1004d88 daab9efc
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710621] 9da0: c1004d88 000188ec daab9ef4 daab9db8 c020c020 c02045a8 56465001 00000120
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710650] 9dc0: ffffffff ffffff00 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710679] 9de0: 00000000 00000000 00000000 00000000 76656c65 7273752f 7665642f 6c756e2f
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710708] 9e00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710738] 9e20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710766] 9e40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710795] 9e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710824] 9e80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710854] 9ea0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710884] 9ec0: 00000000 00000000 40000700 40000700 daab9dd8 00000000 00000054 33c946c8
 
Message from syslogd@raspberrypi at Jul 22 10:30:39 ...
 kernel:[14865.710914] 9ee0: daab9fb0 be90cf68 daab9f8c daab9ef8 c020c100 c020bfb0 00000000 00000240

GramJr, 2019-07-22:
Yes, I have this problem too on rpi 4 (2gb)

richardghirst, 2021-07-02:
No idea what caused this, but it builds and installs fine for me and others.

closed 2021-07-02 by richardghirst