richardghirst/PiBits #116: Installed servoblaster for Raspberry Pi 3 Model B+ but an error occured

stanj98, 2019-10-08:
I've installed servoblaster into my linux system and followed these steps:

- git clone https://github.com/richardghirst/PiBits.git
- cd /PitBits/ServoBlaster/user
- Since '_#include <sys/sysmacros.h>_' under #include <sys/stat.h> is already included in mailbox.c, I installed it right away using: 
- sudo make install

**However, I'm getting this error:**

servod.c: In function ‘get_model_and_revision’:
servod.c:994:6: warning: implicit declaration of function ‘bcm_host_is_model_pi4’; did you mean ‘bcm_host_deinit’? [-Wimplicit-function-declaration]
  if (bcm_host_is_model_pi4()) {
      ^~~~~~~~~~~~~~~~~~~~~
      bcm_host_deinit
servod.c: In function ‘main’:
servod.c:1383:19: warning: implicit declaration of function ‘bcm_host_get_model_type’; did you mean ‘bcm_host_get_sdram_address’? [-Wimplicit-function-declaration]
   int bcm_model = bcm_host_get_model_type();
                   ^~~~~~~~~~~~~~~~~~~~~~~
                   bcm_host_get_sdram_address
/usr/bin/ld: /tmp/cczSzeCZ.o: in function `get_model_and_revision':
/home/pi/PiBits/ServoBlaster/user/servod.c:994: undefined reference to `bcm_host_is_model_pi4'
/usr/bin/ld: /tmp/cczSzeCZ.o: in function `main':
/home/pi/PiBits/ServoBlaster/user/servod.c:1383: undefined reference to `bcm_host_get_model_type'
collect2: error: ld returned 1 exit status
make: *** [Makefile:6: servod] Error 1

Is there something I should include in any file for Rpi 3B+ version?

richardghirst, 2019-10-09:
That's because my recent change to support Pi4B relies on you having a
reasonably up to date install, sorry.

Specifically, your /opt/vc/include/bcm_host.h (and associated library) are
not up to date.  The relevant package is raspberrypi-firmware, and I
have libraspberrypi-dev_1.20190925-1_armhf.deb installed.

Can you apt-get update && apt-get upgrade and see if that helps?  Or grab
the current raspbian SD card image and try with that.

Richard

stanj98, 2019-10-10:
Thank you for your reply, Richard,

I will do the following and get back to you.

Stanley

openiove3690, 2019-10-10:
I have the same problem. Please write me a solution.

openiove3690, 2019-10-10:
sudo rpi-update

I solved this problem.
But be careful, because there may be other problems.

stanj98, 2019-10-10:
How is sudo Rpi update different from sudo apt-get update?

Can you tell me the difference?

stanj98, 2019-10-11:
I tried both the options Richard, but it doesn't work. Do you have any
other ideas?

richardghirst, 2019-10-11:
clever_biscuit,
rpi-update updates the kernel, videocore libraries, etc, to the latest
development version.  apt-get update && apt-get upgrade would update them
(along with everything else) to the latest released versions for the distro
you have installed.  I guess you are having issues because you are still
running jessie, whereas the latest Raspbian images are for buster.

Anyway, an alternative solution might be for you to use the
previous version of servod.c; just grab that from the link below and
replace the servod.c you currently have:

https://github.com/richardghirst/PiBits/raw/4309bab2cae48600e755452a15c89a7ab3f03dfb/ServoBlaster/user/servod.c

Let us know if that works for you,

Richard

stanj98, 2019-10-11:
I will try it out. Sorry for bothering again but how do I replace the code
from the new servod.c to the old servos.c?

Do I cut all code from existing servod file and just copy paste the old
code inside it?

richardghirst, 2019-10-11:
You could click that link in your browser and then right-click and hit
save-as, then use your file manager to move the old servod.c out of the way
and move the saved file in to its place.

Or in a terminal you could do:

cd ServerBlaster/user
mv servod.c servod.c.latest
wget
https://github.com/richardghirst/PiBits/raw/4309bab2cae48600e755452a15c89a7ab3f03dfb/ServoBlaster/user/servod.c

That should fetch the old version of servod.c; then use make as before.

Richard

stanj98, 2019-10-11:
I did as requested but it doesn't seem to work. I'm still getting the error
which I posted above.

Do you have the link to the previous mailbox.c file?

Thanks

richardghirst, 2019-10-11:
The error you posed above was (assuming stanj98 and clever_biscuit are the
same person):

servod.c:994:6: warning: implicit declaration of function
‘bcm_host_is_model_pi4’

That's an issue with servod.c, so why are you asking for mainbox.c?

In any case, that error is related to function ‘bcm_host_is_model_pi4’ and
that function is not referenced in the servo.c that I linked you to.

So, you must have made a mistake saving the old version and copying it in
to place; you are still trying to compile the latest version of servod.c.

stanj98, 2019-10-14:
Oh yeah, you're right. Sorry, I thought of something else. Anyways, I used
the terminal command for replacing servod.c and now the servoblaster is
installed. However, when I run my code, there is no movements at all. What
do you think the reason might be?

richardghirst, 2019-10-15:
The question now is always: is servod broken, or is it your system or
hardware setup?  To figure this out, try the following:

pi@raspberrypi:~ $ sudo killall servod
pi@raspberrypi:~ $ sudo PiBits/ServoBlaster/user/servod

Board model:                     B
GPIO configuration:            P1 (26 pins)
Using hardware:                PWM
Using DMA channel:              14
Idle timeout:             Disabled
Number of servos:                8
Servo cycle time:            20000us
Pulse increment step size:      10us
Minimum width value:            50 (500us)
Maximum width value:           250 (2500us)
Output levels:              Normal

Using P1 pins:               7,11,12,13,15,16,18,22

Servo mapping:
     0 on P1-7           GPIO-4
     1 on P1-11          GPIO-17
     2 on P1-12          GPIO-18
     3 on P1-13          GPIO-27
     4 on P1-15          GPIO-22
     5 on P1-16          GPIO-23
     6 on P1-18          GPIO-24
     7 on P1-22          GPIO-25

pi@raspberrypi:~ $ echo 0=100 > /dev/servoblaster
pi@raspberrypi:~ $ echo debug > /dev/servoblaster
cead9580 cead9940
---------------------------
Servo  Start  Width  TurnOn
  0:      0    100      1
  1:    250      0      0
  2:    500      0      0
  3:    750      0      0
  4:   1000      0      0
  5:   1250      0      0
  6:   1500      0      0
  7:   1750      0      0

Data:
@    0: 0bc60000
@  100: 0bc60010
---------------------------
pi@raspberrypi:~ $

After you echo debug > /dev/servoblaster it first outputs two hex numbers
(cead9580 cead9940 in the example above).  The important things is that
those two numbers are different from each other (but yours will likely be
different from mine, and different every time you run it).  If they are
different it means the DMA controller that servod uses is running ok.

The rest of the output above should be the same for you.  If those first
two numbers are different, and the rest of the data is the same, then it is
most likely servod is working just fine, and the problem is with your setup
- for example, you don't have a common zero volts for the Pi and the
servos, or you have something else running that is driving the gpio or the
PWM hardware that serovd also uses.

You could try running servod with the "--pcm" option and see if that
helps.  That makes it uses the PCM hardware for timing, rather than the PWM
hardware.

If you are still having problems, do the echo debug thing a few times and
send me a list of the two hex numbers it outputs.  That'll let me verify it
seems to be running at the right rate.

Richard

stanj98, 2019-10-16:
Thanks for the detailed information, Richard.
I will get back to you in a while!

tensionhead, 2020-01-05:
I could install servoblaster successfully on Raspbian Jessie on a Pi2 with the link you provided above Richard!

closed 2021-07-02 by richardghirst