richardghirst/PiBits #117: MPU6050 Remote I/O error on RasberryPI 4

paolov, 2019-10-13:
I compiled the MPU6050-Pi-Demo code after changing to /dev/i2c-1 (otherwise I was getting "Failed to open device: Is a directory") and now I'm getting the error "Failed to write reg: Remote I/O error". Here is the log:

```
Initializing I2C devices...
Failed to write reg: Remote I/O error
Failed to write device(-1): Remote I/O error
Failed to write reg: Remote I/O error
Failed to write device(-1): Remote I/O error
Failed to write reg: Remote I/O error
Failed to write device(-1): Remote I/O error
Failed to write reg: Remote I/O error
Failed to write device(-1): Remote I/O error
Testing device connections...
Failed to write reg: Remote I/O error
MPU6050 connection failed
Failed to write reg: Remote I/O error
a/g:   6912      0      0        0      0      0
```
I can access the mpu with c# without problem at address 69 instead of 68 so I tried to initialize the mpu to address MPU6050_ADDRESS_AD0_HIGH but I'm getting the same error.
Here the result of i2cdetect -y 1

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

paolov, 2019-10-13:
Just to confirm that at the end if works perfectly initializing to address 69. It was me making a mistake.
To recap, to make it working on RPi4 you have to change the folder to /dev/i2c-1 and use address accordingly to i2cdetect (69 in my case)

richardghirst, 2019-10-15:
That's good to know, thanks for debugging the problem!