richardghirst/PiBits #118: one pi to another

jasharrar, 2020-01-09:
Hello (not an issue, just a question)

I have a zero running servoblaster and motioneye. Works well, I have set up onscreen buttons to move the camera left and right.
I would like to be able to control this (via terminal etc) from another pi.
I guessed something like sudo /192.168.1.192/ user=pi password=password echo 2=-10 > /dev/servoblaster

but i get permission failed.  having said that even if i remove the ip address I still get that so im defo doing something wrong.  Any ideas?

richardghirst, 2020-01-16:
That "> /dev/servoblaster" is evaluated by the local shell, so tries to
write the output of the preceding command to /dev/servoblaster on the local
machine.

You get permission denied because the local shell is running as a regular
user who does not have permission to write to /dev.

You have to give sudo a command to run, not just an IP address, such as
"sudo ssh pi@192.168.1.192 ...."

The sudo and the output redirection needs to run on the remote system.

Try something like:

ssh pi@192.168.1.192 "sudo echo 2=-10 > /dev/servoblaster"

But, you'll need that to run without it prompting for passwords.... you
could investigate ssh keys and ssh-add to avoid the prompt for remote user
pi's password from the ssh session, and you could investigate sudoers to
figure out how to let the remote user pi run sudo without being prompted
for a password.

On the remote system you could "chmod 777 /dev/servoblaster" so every user
can write to it, and then you wouldn't need the sudo.

Hope that's enough to get you going if you haven't solved it already, sorry
for the slow response!

Richard

jasharrar, 2020-01-17:
Thanks for your help, im much further along with that info. If i work it out ill post my results.

jasharrar, 2020-01-23:
i worked it out!!

so I created the following python file:
from pexpect import pxssh
s = pxssh.pxssh()
if not s.login('192.168.1.195', 'root', 'MYPASSWORD'):
    print("SSH session failed on login.")
    print(str(s))
else:
    s.sendline('echo 2=+10 > /dev/servoblaster')
    print("Right")
    s.logout()

then create the motioneye action button script and save as 'right_1' in /etc/motioneye
#!/bin/bash
/usr/bin/python3 /home/pi/pisys/right_1.py

chmod +X both files
and it works!!  Thanks for your help

jasharrar, 2020-01-23:
also just an an FYI if your trying this there is a 5s delay but not too bad. Tried using rsa keys instead but they drove me mad!!

closed 2021-07-02 by richardghirst