richardghirst/PiBits #119: Bug. Unable to play audio music files when servod is launched.

alhimik1986, 2020-01-22:
Hello.
I found a bug, when servod is launched the music plays very slowly and noisy.
### steps to reproduce the bug:
 - install servod
 - reboot raspberry pi
 - try to play music: omxplayer /home/pi/Desktop/2.mp3
 - the music will play slowly and noisy

### next steps:
 - find servod process: ps -A | grep servod
 - kill servod process: sudo kill <servod process id>
 - try to play music: omxplayer /home/pi/Desktop/2.mp3
 - the music will play well

I ask you to fix the bug not because I like listening music. I try to transmit the sound to raspberry and manipulate the servos at same time. Advance grateful.

richardghirst, 2020-01-22:
Have you tried invoking servod with the --pcm option?

alhimik1986, 2020-01-27:
Yes. It's helped, thanks!

richardghirst, 2021-07-02:
Closing as issue resolved

closed 2021-07-02 by richardghirst