richardghirst/PiBits #123: is it important that /proc/cpuinfo for the Pi Zero varies by the OS version ?

NewtownGal, 2020-10-29:
I'm trying to solve a problem compiling servod.c on Buster and discovered that /proc/cpuinfo, which servod uses to identify the processor, varies depending on the OS version. With old Wheezy and a Pi Zero, /proc/cpuinfo gives this:

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 900093
Serial          : 00000000d5f5cdb8

Buster, also with a Pi Zero, gives this:

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 900093
Serial          : 0000000078854fc8
Model           : Raspberry Pi Zero Rev 1.3

The hardware version, BCM2708 vs BCM2835, is used by servod. I have ServoBlaster running, apparently ok, on Wheezy machines that report BCM2708. That version of servod supposedly checks for both models. I cannot compile on Buster on a Pi Zero due to a problem with bcm_host.h, so I don't know how it behaves.

What is the effect of this ? If necessary, can I use the version of ServoBlaster that was compiled for Wheezy work on Buster if I change the /proc/cpuinfo on Buster to be the same as Wheezy ? Or, can someone supply me servod that was compiled for Buster, or at least something newer than Wheezy ?

richardghirst, 2021-07-01:
Yes, it was important as the code used to look at the "Hardware:" string to make decisions. I just delivered a change to base everything off the "Revision:" string instead, so the code should now work on the Pi Zero and elsewhere regardless of OS version.  I'll grab a recent OS image next, to see if the build issue you had still needs resolving.

richardghirst, 2021-07-02:
Closing as the dependency on /proc/cpuinfo "hardware" string is removed.  Will address building under Buster via issue #122.

closed 2021-07-02 by richardghirst