richardghirst/PiBits #125: brushless motors run few seconds

peniblou, 2020-11-27:
I am building a ROV (remotely operated underwater vehicle). My first prototype was running with 2 brushless motors and I used the PWM function of the Raspberry pi 3 by installing a library  http://librpip.frasersdev.net/peripheral-config/pwm0and1/  It works perfectly.  

Then, few months ago, for my second prototype (3 brushless motors), I have installed ServoBlaster following the below procedure :
git clone https://github.com/richardghirst/PiBits.git 
cd PiBits/ServoBlaster/user
sudo make
An error occurs….. Then...

sudo apt-get update && apt-get upgrade  # (following the issue 116 recommandation)
sudo rpi-update  
sudo reboot

cd PiBits/ServoBlaster/user   
sudo make
su
sudo make install

echo 0=100 > /dev/servoblaster 
echo 0=110 > /dev/servoblaster 
It was working perfectly…

My problem: 
I have reinstalled ServoBlaster following the same procedure few weeks ago, and now when I issue the commands 
echo 0=100 > /dev/servoblaster 
echo 0=110 > /dev/servoblaste
The brushless motor just run few seconds and the stops. 
I have to reissue the same commands to make it run again …. few seconds. 

By investigating, I found out that the following procedure makes it run .  Of course, I do not feel completely confident in this !!!
cd /home/pi/PiBits/ServoBlaster/user
sudo ./servod
echo 0=100 > /dev/servoblaster 
echo 0=110 > /dev/servoblaster
It works !!!   but after a reboot, I have to issue the same command sudo ./servod to make it run. 

Remark : For information, I have also installed a joystick SONY dualshock3 bluetooth, I do not believe it can interfere 

Have you any idea how I can fix it properly ? 

Many thanks for your help

Michel

richardghirst, 2020-11-27:
Sounds like you are running servod with the --idle-timeout option, which is
intended to avoid servo motors overheating when they are not being used.

You'll see this line in the /etc/init.d script:

OPTS="--idle-timeout=2000"

Change it to

OPTS=""

Richard

peniblou, 2020-12-01:
In the file /etc/init.d/servoblaster  , I changed to OPT=""  and it works perfectly ! 

Last question :  In order to make it run, I have to issue the commands 
cd /home/pi/PiBits/ServoBlaster/user
sudo make install

I would like servo to start at boot : I created a startup script with these 2 commands in the /etc/init.d directory. It does not work...   How do you recommend to run these commands ?  In a new file, in an existing file (like servoblaster) ? 

Many thanks for your help

Michel

richardghirst, 2020-12-01:
"sudo make install" is supposed to copy the init-script to
/etc/init.d/servoblaster and then run update-rc.d such that the init script
runs automatically on reboot.  It sounds like that isn't happening for you.

You should see symlinks in /etc/rc2.d, etc, to /etc/init.d/servoblaster,
which are invoked to start servod as the system boots.

What output do you get from running "sudo make install" and "ls -l
/etc/rc?.d/*servoblaster"?  something like this?  Any errors or warnings
output?

pi@raspberrypi:~/ServoBlaster $ sudo make install
[ "`id -u`" = "0" ] || { echo "Must be run as root"; exit 1; }
cp -f servod /usr/local/sbin
cp -f init-script /etc/init.d/servoblaster
chmod 755 /etc/init.d/servoblaster
update-rc.d servoblaster defaults 92 08
/etc/init.d/servoblaster start

pi@raspberrypi:~/ServoBlaster $ ls -l /etc/rc?.d/*servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc0.d/K01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc1.d/K01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc2.d/S01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc3.d/S01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc4.d/S01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc5.d/S01servoblaster ->
../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Dec  1 19:01 /etc/rc6.d/K01servoblaster ->
../init.d/servoblaster
pi@raspberrypi:~/ServoBlaster $

Which OS are you running (i.e. what image did you write to your sdcard)?

You could probably work round the problem by adding this line to
/etc/rc.local, just before the "exit 0" line at the end:

/etc/init.d/servoblaster start

but that shouldn't be necessary.

Richard

peniblou, 2020-12-04:
Bonjour Richard,

This is below what I got : 

pi@raspberrypi:~/PiBits/ServoBlaster/user $ sudo make install
[ "`id -u`" = "0" ] || { echo "Must be run as root"; exit 1; }
cp -f servod /usr/local/sbin
cp -f init-script /etc/init.d/servoblaster
chmod 755 /etc/init.d/servoblaster
update-rc.d servoblaster defaults 92 08
/etc/init.d/servoblaster start
pi@raspberrypi:~/PiBits/ServoBlaster/user $ ls -l /etc/rc?.d/*servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc0.d/K01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc1.d/K01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc2.d/S01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc3.d/S01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc4.d/S01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc5.d/S01servoblaster -> ../init.d/servoblaster
lrwxrwxrwx 1 root root 22 Nov 21 17:20 /etc/rc6.d/K01servoblaster -> ../init.d/servoblaster
pi@raspberrypi:~/PiBits/ServoBlaster/user $ 

Which looked like OK. 

 Then I edited etc/rc.local and I found the following line before "exit 0"
/usr/local/bin/librpip-util/librpip-pwm-init. 
I deleted this line and Servoblaster is now starting at boot !  Problem solved !

As a matter of fact, for my first prototype, I previously installed a PWM library and used the PWM function of my Raspberry. 
see (https://librpip.frasersdev.net/peripheral-config/pwm0and1/)

Looks like this library and Servoblaster cannot be run simultaneously. Since I only need Servoblaster, I deleted this line...

Richard, again many thanks for your help, ServoBlaster is definitely a great tool and I would recommend it to my friends,

 Michel

peniblou, 2020-12-04:
Everything is working fine. 
Thanks
Michel

closed 2020-12-04 by peniblou