richardghirst/PiBits #127: modprobe: FATAL: Module servoblaster not found in directory /lib/modules/5.4.79-v7l+

samaldude, 2020-12-27:
Sorry if this turns out to be a question from a newbie rather than an issue, but I get a FATAL at the end following what I believe is the correct steps.

OS: Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l
Hardware: Raspberry Pi 4 Model B Rev 1.2
Steps:
1. sudo apt-get update && sudo apt-get install git python-opencv python-all-dev libopencv-dev
2. git clone https://github.com/richardghirst/PiBits.git
3. cd PiBits/ServoBlaster/usr; sudo make install
4. Modify servod.c to add (strstr(modelstr, "BCM2711")
5. sudo modprobe servoblaster
Error: 
modprobe: FATAL: Module servoblaster not found in directory /lib/modules/5.4.79-v7l+

richardghirst, 2020-12-27:
The current version of servoblaster runs as a userspace process, not as a
kernel module.  You've built the userspace servod process, and you can
either start it up manually via

sudo ./servod

Or you can install it such that it is started automatically on system
boot, via

sudo make install

Hope that helps; the original version was a kernel module and the
documentation has sections for both.  I should probably delete the kernel
module version as it is unmaintained.

Richard

samaldude, 2020-12-30:
Thank you that solved my problem. A different issue related to adafruit tft 2.8 resistive display. As soon as servod is invoked display goes dark.  If dma channel 14 is used instead of 7 then display remains but no servo reactions is observed.

richardghirst, 2020-12-30:
Have you experimented with the pcm/pwm options? That may or may not help.

samaldude, 2020-12-31:
Actually I must have not used sudo the first time and not noticed the error.  But tried both channels 7/14 and pcm/pwm combinations, they all make the screen go black. The screen is not using the GPIO pins 4/17 so not sure what the interaction is, but its OK don't waste your time if you are not interested yourself, I will report back if I pursue to figure it out. Thanks again.

richardghirst, 2021-07-02:
Closing this issue as servoblaster itself works and I don't have the hardware to investigate interactions with the Adafruit display, sorry!

closed 2021-07-02 by richardghirst