richardghirst/PiBits #128: Me Arm Raspberry Pi issue

CP101, 2021-01-19:
Many years ago I managed to get the template for the MeArm (Robotic Arm) that used several servo motors to move around
I work for a school and we can lazer cut the template, buy some cheap servo motors and connect directly to the raspberry pi

In the past I have done similar to the following commands

cd PiBits/ServoBlaster/user
make servod
sudo ./servod --idle-timeout=2000
echo 0=50% > /dev/servoblaster

When a new version of Raspbian comes out I have bownloaded the updates here.
When i try the current code on a version of Raspbian that was out before PI4 (Stretch) it give me a error about its not a PI4
I have tried the latest and the 2 older versions of busted that must have been current 16 months ago when this was last updated, but the servo movements dont seem reliable or dont work at all.
[MeArm.zip](https://github.com/richardghirst/PiBits/files/5836875/MeArm.zip)

2019-04-08-raspbian-stretch
2019-06-20-raspbian-buster
2019-07-10-raspbian-buster-full
2020-02-14-raspbian-buster

I have attached the code i am trying to get working, Im not a coder so not sure if there is something that has changed in the last year or 2.

Any help with any past would be greatly appreciated.
I have the template and build instruction availabale if anyone wants to make a mearm for less than £10.

CP101, 2021-01-20:
I suppose what I am asking in short is, is there a version of servo blaster and raspbian that works best together.
For me its easy to reinstall the OS to whatever version :)

richardghirst, 2021-01-20:
Which version of Pi hardware are you using?

CP101, 2021-01-20:
I have been using a 3b+ mostly, but have tried some on a 3b as well.
Some of the students are using 2b.

Thank you so much for getting back to me, you and elon musk are my heroes.

richardghirst, 2021-01-20:
OK, I believe there are issues with the Pi Zero and Pi4 using the current
servoblaster code and recent versions of the OS but I'm not aware of issues
with the 3b+.

Let's stick with the 3b+, the current servoblaster code, and
2020-02-14-raspbian-buster for now.

Also let's stick with doing things from the shell prompt to keep things
simple.  Try this:

cd PiBits/ServoBlaster/user
make servod
sudo killall servod # to make sure there isn't already one running
sudo ./servod --idle-timeout=2000

echo 0=0% > /dev/servoblaster
echo 0=100% > /dev/servoblaster
echo 0=0% > /dev/servoblaster
echo 0=100% > /dev/servoblaster
...

Paste the output of the commands here - there should be a bunch of
information printed out when you start servod.

What happens when you do the above?  Does the servo for the "Waist" (I
guess) move between minimum and maximum range as expected?

You can also do:

echo debug > /dev/servoblaster

That will output a load of data; the most important bit is the first line,
with two hex numbers on it.  The numbers should be different; indicating
that the DMA engine is running.

Sometimes issues are caused by power problems, such as no common 0 volts
between the Pi and the servo power, for example.
Sometimes there is other software running that is using the same hardware
as servoblaster, and running it with a --pcm or --dma-chan=7 parameter can
help.
Sometimes some other software has been installed that uses GPIO pins and
interferes with servod.
Also of course the servos can be faulty, the GPIO outputs on the Pi can be
damaged, or the servos can be wired up wrongly, but I assume you've got the
problem on more than one setup which should eliminate those options.

If no servo movement happens and you have access to an oscilloscope or
logic analyser you could disconnect the servos and probe the GPIO pin to
see if you get a train of pulses for a couple of seconds after setting the
servo position.  The pulses will stop after 2 seconds due to the idle
timeout parameter.

Let me know what happens in the above tests, and we'll take it from there.

Richard

CP101, 2021-01-20:
Thank you for your help.

I have now tested on PI2B, 3B,and 3B+
Using a fresh install of 2020-02-14-raspbian-buster on each
Running the commands as listed 
[image]
This is my setup
[image]
I ran the debug command
[image]
We have 12 teams of 2, here are some pics of their setups
[image]
[image]

I got some fresh servo motors and tested they had full range of movement on our original proof on concept, that has a ardeno attached
[image]
I also used fresh batteries.

After the first 2 motors vibrating the first time i set them from 0 to 100% and back again, all motors seem to have full range of movement and the code i have now works with nothing attached, now thats solved its time to figure out what the students have done wrong.
Thank you for shining a light on the problem, I can now see a way to get the kids motivated again, they where disheartened when it would not work like the demo arm.

Here is the code, the accembily instructions, and the template in 2d design and auto cad, incase you wanted to give it ago
[MeArm.zip](https://github.com/richardghirst/PiBits/files/5843518/MeArm.zip)
The base plate is 144mm long and 95mm wide incase you need scale to base the size of off like i did when converting in to a usable format.

richardghirst, 2021-01-20:
I'm glad you've made progress, but in your photos I cannot see any
connection between the 0 volts on the Pi to the 0 volts on the battery
pack.  I assume the four wires from the GPIO header are the four servo
control signals.
I assume you are powering the Pi from a regular USB supply, and the servos
from the battery pack.  You need to join the 0 volts from the battery pack
and the 0 volts from the Pi, because otherwise the servos have no reference
for the control signal from the Pi.  Without that connection you will get
erratic behaviour.  There are a few "0v" pins on the Pi GPIO header;
connect one of them to the 0v of the battery pack and it should work a lot
better.

Richard

richardghirst, 2021-07-02:
Closing this as it seems the issue was resolved and the most recent update should work on all Pi models and recent OS versions.

closed 2021-07-02 by richardghirst