richardghirst/PiBits #129: complie issue of Pi4 8GB

mrprohde, 2021-06-04:
Having issues when I try to make the SW.  (running ubuntu in an LXD container on a Pi4 running ubuntu as the host OS.

ubuntu@testros:/home/ubuntu/git/PiBits/ServoBlaster/user$ make
gcc -Wall -g -O2 -L/opt/vc/lib -I/opt/vc/include -o servod servod.c mailbox.c -lm -lbcm_host
servod.c:44:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
mailbox.c: In function ‘mapmem’:
mailbox.c:69:33: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
       printf("mmap error %d\n", (int)mem);
                                 ^
Makefile:6: recipe for target 'servod' failed
make: *** [servod] Error 1

ZalgoSoft, 2021-07-01:
root@raspberrypi:~# rpi-source
root@raspberrypi:~/PiBits/ServoBlaster/kernel# make -C /root/linux/ M=`pwd` modules

richardghirst, 2021-07-02:
@ZalgoSoft looks like you are trying to build the kernel module version of servoblaster.  That is no longer supported, and you should use the userpsace version, as per the README.

@mrprohde I just checked, and it builds fine on the latest Raspberry Pi OS image (Buster, desktop version).  I don't run Ubuntu on my Pi and I have no idea whether the Ubuntu image you have in your container includes the bcm_host library needed to query various Pi hardware addresses and the like.  It looks like it does not.  It also looks like, from the warning on mailbox.c, like you are running a 64 bit OS, and servoblaster is currently only supported on 32 bit Raspberry Pi OS.

closed 2021-07-02 by richardghirst