richardghirst/PiBits #130: Servoblaster does not work well on new Raspberry installation

Igor-Shabalin, 2021-06-18:
Hi
Such a problem. I installed a new system Buster on a new SD card. Servoblaster only works in the 500-1500 range. Doesn't work after reboot. It is started only manually by commands . 
sudo /etc/init.d/servoblaster stop
sudo /etc/init.d/servoblaster start
On the same hardware with an old SD card, everything works well in 500-2500 range
Any ideas ?

richardghirst, 2021-07-02:
You don't say what model of Pi you are using or what OS versions are on the old and new SD cards.  However, there was a problem with Pi Zero, where it stopped working properly with newer OS versions.  I've just fixed that, so could you try the latest version of servod.c?

Igor-Shabalin, 2021-07-02:
I fixed it.

Not work on:
Raspberry Pi OS with desktop and recommended software
Release date: May 7th 2021

But work on : 
Raspberry Pi OS Lite
Release date: May 7th 2021

richardghirst, 2021-07-03:
If you have an HDMI display connected running a graphical desktop I found that you need to invoke servod with the --pcm option which you can do by editing /etc/init.d/servoblaster and modifying the OPTS string, so it looks like this:

OPTS="--idle-timeout=2000 --pcm"

Having done that I found servoblaster worked OK following a reboot, using the desktop OS with recommended software.

Closing this issue as I consider it resolved.

closed 2021-07-03 by richardghirst

Igor-Shabalin, 2021-07-03:
I don't use displays with raspberry, only ssh. But now everything is working as it should. Today I tried it with Pi4. Works too. Thank you.